realsenseai/librealsense #3581: Does D435i require calibration to use with OpenCV?

npd26, 2019-03-25:
Hello,

I am using Realsense D435i with OpenCV, extracting only  RGB image for object POSE estimation. In general to use pose estimation algorithm in OpenCV it requires Camera Matrix and Distortion coefficient matrix. So, How can I get this matrix information to use with OpenCV?
or should I perform calibration for D435i in OpenCV as standard pinhole camera calibration procedure?

Thanks.

dorodnic, 2019-03-25:
Hi @npd26
In librealsense these coefficients are stored inside `rs2_intrinsics` struct and you can query it from the API:
[get-video-stream-intrinsics](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-video-stream-intrinsics)

npd26, 2019-03-25:
thank you. so same as depth I can get it for color stream.

npd26, 2019-03-25:
Hi @dorodnic ,
sorry I am not getting it how to use above rs2_intrinsics with this code,

````
int main(int argc, char * argv[])
{
`
	// Declare RealSense pipeline, encapsulating the actual device and sensors
	rs2::pipeline pipe;
	
	//Create a configuration for configuring the pipeline with a non default profile
	rs2::config cfg;

	//depth stream config
	//cfg.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_Y8, 30);
	//cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);//1280,720
	
	//colour stream config
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
	
	
	// Start streaming
	pipe.start(cfg);

	while (true)
	{
		// Camera warmup - dropping several first frames to let auto-exposure stabilize
		rs2::frameset frames;
		
		frames = pipe.wait_for_frames();

		//Get each frame
		rs2::frame color_frame = frames.get_color_frame();
	    
		// Creating OpenCV Matrix from a color image
		Mat color(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
		
		// Display in a GUI
		namedWindow("Display Image", WINDOW_AUTOSIZE);
		imshow("Display Image", color);
		
		if (waitKey(30) == 'q')
		{
			return 0;
		}	
	}
}`

dorodnic, 2019-03-25:
Once you have `color_frame` you can call `color_frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics()`

closed 2019-03-25 by npd26