realsenseai/librealsense #3582: D435i is recognized as USB2.1,anyone could tell me what happened

Dragon1998, 2019-03-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   | 5.1.4 | 
| Operating System & Version |   Linux (Ubuntu 16 ) |

### Issue Description
I want use the D435i camera,it is recognized as USB2.1.I don't know why can you help me!I am crazy~~

MartyG-RealSense, 2019-03-25:
Could you please first try plugging the camera into the USB port quickly and firmly, as a slow insertion may cause it to be recognized as USB 2.1 instead of USB 3.

Dragon1998, 2019-03-27:
It still not work.It is still recognize as USB2.1

Dragon1998, 2019-03-27:
Alright,you are right it work now!Thank you very much!!!

dorodnic, 2019-03-27:
Awesome, thank you @MartyG-RealSense

closed 2019-03-27 by dorodnic