realsenseai/librealsense #3583: question about d435 measurement accuracy

Cavendish-Koo, 2019-03-25:
| Camera Model                       | { D435 } | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  16.04  | 
| Platform                                 | NVIDIA Jetson TX2 |

### Issue Description
Hi,guys.Rencently I want to calculate the camera's current yaw angle from two points on a horizontal line, but I found that the error is large at a distance and the calculated yaw angle fluctuates greatly. 

So I hope to reduce the error by least squares, but I find that the fluctuations become bigger. When I output these points, I found an interesting phenomenon. The depth value of each point is incremented by 5mm. 
[image]
[image]

I know the accuracy of this camera is millimeter, but I have never heard of this phenomenon. I doubt if this is related to the filtering inside the camera.

I also use the function rs2_deproject_pixel_to_point(),but the continuity in the x direction illustrates its correctness.

In addition, this phenomenon has nothing to do with the angle of the camera, that is, the depth step of each adjacent point is never near 5mm.

agrunnet, 2019-03-27:
@Cavendish-Koo 
There are two aspects to your question as I see it:
 
1. Why is the step size at 1.8m about 5mm and not 1mm?  The reason for this is that we calculate depth to resolution of 1/32 of a disparity . So we have 32 subpixels in the calculation. There is an inverse relationship between depth and disparity. For the D435 at 848x480, the subpixel steps-ize becomes ~5mm.   At 1m the fundamental resolution limit of a single point is 1.5mm, and at 20cm it is 60um. 
For the D415 this is 2.6x better at 1280x720.   
  
2. Why is the curve not exactly linear? This is a different phenomenon and actually something we are working on. We may have a fix for this soon, and it will be released as a FW upgrade.

Cavendish-Koo, 2019-03-28:
Thank you for your answer. I will wait for your good news to fix this problem. 

I will close this question, thank you again.

closed 2019-03-28 by Cavendish-Koo