realsenseai/librealsense #3584: How Can I Achieve Depth&Color Alignment Using the New Android Wrappers?

mikaellaine, 2019-03-25:
The new Anroid wrappers for non-rooted devices allows streaming depth and color, but don't seem to allow alignment (as rs2::align). Is there some way around this, or is this a feature still in development?

dorodnic, 2019-03-26:
Hi @mikaellaine
We are gradually adding more APIs to the JNI wrapper, but the bigger issue is that current rs2::align implementation is likely to be very slow on a phone (it is fairly CPU heavy and optimized only for x86 / CUDA) Translating every pixel using functions similar to `rsutil.h` from RenderScript would probably be a better solution for now.

mikaellaine, 2019-03-26:
Thank you for your reply @dorodnic 
I have indeed used the rs2::align API on a Samsung Tab S3 Android tablet and the framerate is around 1.5 so pretty slow. We'll see if simply transforming points in 3D is enough or faster for our purposes. Thanks again!

Murded, 2019-03-27:
@mikaellaine Would be intrested on how much quicker this was on the same device?

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@mikaellaine Any other questions for this ticket? Looking forward to your update. Thanks!

closed 2019-05-17 by RealSenseCustomerSupport