realsenseai/librealsense #3586: Firmware 5.11.6

tRosenflanz, 2019-03-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 5.11.1.100 |                        | 

### Issue Description
Hello, 

In the release notes for SDK 2.19.1 firmware 5.11.6 is mentioned as required for D435i however the website [only has 5.11.1.100 and 5.11.4](https://realsense.intel.com/intel-realsense-downloads/). Should we be looking for 5.11.6 somewhere else?

dorodnic, 2019-03-26:
Hi @tRosenflanz
Please note that the firmware for D435i stated in 2.19.1 release notes is 5.11.4.0+.
The firmware required for **D430i** (different camera) is in fact 5.11.6 and it will be released shortly (firmware release process is somewhat more complicated, but we are trying to make sure everything is preemptively supported in librealsense)

tRosenflanz, 2019-03-26:
Sounds good. I missed the D430i detail. Hopefully 5.11.4 works well (having problems with 5.11.1.100 so far)

closed 2019-03-26 by tRosenflanz