realsenseai/librealsense #3587: Unable to launch "realsense_camera zr300_nodelet_rgdb.launch"

MangoFighter87, 2019-03-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | ZR300 | 
| Camera Firmware Version     | 2.0.71.28 | 
| Operating System & Version | Ubuntu 16.04 LTS | 
| Kernel Version (Linux Only)    | 4.15.0-46-generic | 

### Issue Description
Was unable to launch **realsense_camera zr300_nodelet_rgdb.launch** due to this error:

<pre>
[zr300_nodelet_rgdb.launch] is neither a launch file in package [realsense_camera] nor is [realsense_camera] a launch file name
The traceback for the exception was written to the log file
</pre>

How do I solve this issue? Your help is very much appreciated, thank you!

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@MangoFighter87  What's the RealSense ROS version for this issue? And please note that ZR300 is EOL(End Of Life) product, the support will be a bit limited now.

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@MangoFighter87 Any update? Thanks!

MangoFighter87, 2019-04-09:
Now able to launch the ZR300 nodelet as there was a typo.. rgdb.launch was changed to rgbd.launch and it worked.. Thanks!

closed 2019-04-09 by MangoFighter87