realsenseai/librealsense #3589: Z-axis orientation in PLY file

ynakayji, 2019-03-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |2.19.0                          | 
| Language                            |  C++                         | 

Hi,
I set D435i toward the ceiling from my desk and exported .ply file by RealSense Viewer, shown below:
[image]

I checked the file with MeshLab and found the Z values of the points were negative, shown below:
[image]

Is it correct behavior ?

Thanks.

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi @ynakayji,
Thank you.
This is expected as starting from SDK 2.19.0, the points are converted from right hand to left hand coordinate system when exporting vertices to ply.
https://github.com/IntelRealSense/librealsense/pull/3352/commits/9e4b6ba09ac6fd0b20bc28f388e44791f4cc04c1

ynakayji, 2019-04-11:
Hi @RealSenseCustomerSupport ,
Thank you for your reply.
Could you tell me the reason why the coordinate system has been changed from SDK 2.19.0.
The coordinate system displayed on 3D RealSense Viewer v2.19.0 seems to remain right hand one.

aangerma, 2019-05-01:
Hi @ynakayji,
The reason we changed the coordinate system when exporting vertices to ply is that when users loaded it with MashLab its looked flipped because the default view point of Mesh Lab is different from our view point on 3D RealSense Viewer and thus we should also changed the faces normals.  So we adjust it to the default definition of MeshLab.

ynakayji, 2019-05-07:
Hi @aangerma ,
Thanks for your reply.
I understood the reason.

closed 2019-05-07 by ynakayji