realsenseai/librealsense #3590: Save aligned color and depth from bag to ply D415

vik-timmersion, 2019-03-26:
I use windows 10, Intelrealsense D415
I have a problem to obtain the aligned color and depth in ply file from bag.

I recorded a bag file.

Then I align each frame of bag file and export to ply, but ply file shows misaligned depth and color.

Could you explain how to save to the ply file from bag file ?

Bag file was recorded the following streams:

	cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
	cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);
	cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_RGB8, 30);

My code is :

void alignDepthToColorFromSeq(std::string pathBag)
{
	// Create and initialize GUI related objects
	window app(1280, 720, "CPP - Align Example"); // Simple window handling
	ImGui_ImplGlfw_Init(app, false);      // ImGui library intializition
	rs2::colorizer c;                          // Helper to colorize depth images
	texture renderer;                     // Helper for rendering images

	rs2::config config;
	config.enable_device_from_file(pathBag);

	// Create a pipeline to easily configure and start the camera
	rs2::pipeline pipe;
	//Calling pipeline's start() without any additional parameters will start the first device
	// with its default streams.
	//The start function returns the pipeline profile which the pipeline used to start the device
	rs2::pipeline_profile profile = pipe.start(config);

	// Each depth camera might have different units for depth pixels, so we get it here
	// Using the pipeline's profile, we can retrieve the device that the pipeline uses
	float depth_scale = get_depth_scale(profile.get_device());

	//Pipeline could choose a device that does not have a color stream
	//If there is no color stream, choose to align depth to another stream
	rs2_stream align_to = find_stream_to_align(profile.get_streams());

	// Create a rs2::align object.
	// rs2::align allows us to perform alignment of depth frames to others frames
	//The "align_to" is the stream type to which we plan to align depth frames.
	rs2::align align(align_to);

	// Define a variable for controlling the distance to clip
	float depth_clipping_distance = 1.f;
	bool stop = false;
	bool playing = false;
	// get playback device and disable realtime mode
	auto playback = (profile.get_device()).as<rs2::playback>();
	playback.set_real_time(false);
	rs2::pointcloud pc;
	rs2::points points;
	int i = 0;
	while (!stop & app) // Application still alive?
	{
		// Using the align object, we block the application until a frameset is available
		rs2::frameset frameset;
		if (pipe.poll_for_frames(&frameset))
		{
			if (profile_changed(pipe.get_active_profile().get_streams(), profile.get_streams()))
			{
				//If the profile was changed, update the align object, and also get the new device's depth scale
				profile = pipe.get_active_profile();
				align_to = find_stream_to_align(profile.get_streams());
				align = rs2::align(align_to);
				depth_scale = get_depth_scale(profile.get_device());
			}

			//Get processed aligned frame
			auto processed = align.process(frameset);

			// Trying to get both other and aligned depth frames
			rs2::video_frame other_frame = processed.first(align_to);
			rs2::depth_frame aligned_depth_frame = processed.get_depth_frame();
			//If one of them is unavailable, continue iteration
			if (!aligned_depth_frame || !other_frame)
			{
				continue;
			}

			// Tell pointcloud object to map to this color frame
			auto color_frame = processed.get_color_frame();
			pc.map_to(color_frame);
			points = pc.calculate(aligned_depth_frame);
			i++;
			std::string pathPly = "D:/Z/Register/RecordTest/viktoriia_rgb1280x720_depth1280x720_disparityShift_0/frame_" + std::to_string(i);
			pathPly += ".ply";
			points.export_to_ply(pathPly , color_frame);
			
			// Passing both frames to remove_background so it will "strip" the background
			// NOTE: in this example, we alter the buffer of the other frame, instead of copying it and altering the copy
			//       This behavior is not recommended in real application since the other frame could be used elsewhere
			remove_background(other_frame, aligned_depth_frame, depth_scale, depth_clipping_distance);
			// Taking dimensions of the window for rendering purposes
			float w = static_cast<float>(app.width());
			float h = static_cast<float>(app.height());

			// At this point, "other_frame" is an altered frame, stripped form its background
			// Calculating the position to place the frame in the window
			rect altered_other_frame_rect{ 0, 0, w, h };
			altered_other_frame_rect = altered_other_frame_rect.adjust_ratio({ static_cast<float>(other_frame.get_width()),static_cast<float>(other_frame.get_height()) });

			// Render aligned image
			renderer.render(other_frame, altered_other_frame_rect);

			// The example also renders the depth frame, as a picture-in-picture
			// Calculating the position to place the depth frame in the window
			rect pip_stream{ 0, 0, w / 5, h / 5 };
			pip_stream = pip_stream.adjust_ratio({ static_cast<float>(aligned_depth_frame.get_width()),static_cast<float>(aligned_depth_frame.get_height()) });
			pip_stream.x = altered_other_frame_rect.x + altered_other_frame_rect.w - pip_stream.w - (std::max(w, h) / 25);
			pip_stream.y = altered_other_frame_rect.y + (std::max(w, h) / 25);

			// Render depth (as picture in pipcture)
			renderer.upload(c.process(aligned_depth_frame));
			renderer.show(pip_stream);

			//// Using ImGui library to provide a slide controller to select the depth clipping distance
			ImGui_ImplGlfw_NewFrame(1);
			render_slider({ 5.f, 0, w, h }, depth_clipping_distance);
			ImGui::Render();
			playing = true;
		}
		else
		{
			using namespace std::chrono_literals;
			std::this_thread::sleep_for(10ms);
		}

		stop = playing && (playback.current_status() == RS2_PLAYBACK_STATUS_STOPPED);
	}
}

Next step, I want to use the resolutions for color 1920x1080 and for depth 1280x720

vik-timmersion, 2019-03-27:
Finally, I find that I do not need to give the aligned depth and color.
I just need to give 
frameset.get_color_frame()
frameset.get_depth_frame()

closed 2019-03-27 by vik-timmersion

nevynwong, 2019-06-25:
Hi vik-timmersion ,
I have a similar problem. I want to save the aligned color and depth frames. I don't know how to do it. Can you help me? Thanks.