realsenseai/librealsense #3592: T265: Allow defining a mask for excluding parts of robot?

skohlbr, 2019-03-26:
Depending on the type of robot used and various real-world constraints it might not be possible to mount a T265 on a robot in a way that that avoids robot parts to be visible in the camera view(s). While the SLAM system on the compute module can be expected to reject implausible feature matches due to static elements wrt to the camera to some extend, a more principled way to go about this would be to be able to make an API available that allows providing a mask, excluding certain image regions. This could probably be reasonably approximated using a single image as opposed to one for each camera. In extreme cases, updating this mask in real-time would even be useful (let's say an arm can move in front of the camera), but that's pretty far out I guess :)

Not sure this is even possible to be added at all, but I wanted to have this mentioned as it's relevant for quite a few real-world applications.

SlavikLiman, 2019-03-27:
Thanks for the suggestion, @skohlbr .
Nevertheless, thanks to the wide FOV, moving robot parts in camera FOV shouldn't confuse it and the T265 tracking camera should continue to report correct position, providing sufficient visibility of static tracked visual features is available.

closed 2019-04-12 by RealSenseCustomerSupport

xpetresx, 2019-07-09:
Dear developers,
I would like to point out, that the masking feature would be very handy. In our particular application that concerns underground operations of mobile robots, when the robot carries its own lightsource. The image from T265 can look like follows:
[image]

As the front legs of the robot are visible in the FOV, the localization persistently goes nuts after awhile. On the other hand, when the legs are covered such that the camera doesn't see them, it can provide flawless localization for hundreds of meters long trails in the underground areas (tunnels, caves, ...). However, shielding the legs from the camera FOV mechanically is very bad for the locomotion capabilities of the robot, and mounting the camera elsewhere on the robot is also out of question. Therefore, a simple masking with a static mask would be very helpful and we would greatly appreciate such a feature.