realsenseai/librealsense #3594: Is there an option to enable compression as it writes to disk?

beomsik-landingai, 2019-03-27:
Required Info |  
-- | --
Camera Model | D435i
Firmware Version | 5.11.1
Operating System & Version | Win10
Platform | PC
SDK Version | 2.19.1
Language | c++

I wanted to save a compressed bag file while a camera is recoding, and it seems it's supported by using recorder with compression_enabled with true. But I couldn't find any example that uses recorder directly. Only example I see that references recorder is playback-record one which uses enable_record_to_file API through config. But I couldn't figure out how pipeline works with a new recorder instance as it doesn't seem there's any api that passes a device to pipeline.

Can any of you point out where I can find similar example or share some code snippet that records with compressed bag? Also, it will be nice it shows how to set a compression level.

Thanks in advance.

MartyG-RealSense, 2019-03-27:
Version 2.17.0 of the SDK introduced a new API to control recording compression.

https://github.com/IntelRealSense/librealsense/pull/2687

A function added to the SDK's API called **rs2_create_record_device_ex** creates a recorder and explicitly enables or disables compression.  By default, compression will be enabled based on device type.  D435i and T265 devices that provide high FPS streams disable compression by default to avoid frame drops during recording.

https://github.com/belkinirena/librealsense/blob/9d8d3a690ebd5fd4b60c3d3c73a6e11e901f14bc/include/librealsense2/h/rs_record_playback.h#L49

dorodnic, 2019-03-27:
Looking at the API changes, I don't think it is currently exposed via pipeline config method. 
If you are building from source, the easiest way to hack it would be to change this line:
[`ds5-factory.cpp:475`](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-factory.cpp#L475)
Please note that this is only ROS-bag lossless LZ4 compression, so don't expect significant file-size reduction. It might, however, cause frame drops on high frequency streams, this is why we disabled it for the D435i.

beomsik-landingai, 2019-03-27:
Thanks for quick response both of you. Appreciated.

@dorodnic got it, I'm having some difficulties that building library from source directly, but that will be another issue. Would it be possible to control these through pipeline config method in the near future?

@MartyG-RealSense so it sounds like pipeline doesn't support that. Do you have any example code that records without using pipeline? I also need to set resolution and framerate without using pipeline in that case.

MartyG-RealSense, 2019-03-27:
@dorodnic or another member of the Intel team will be best equipped to answer these questions, as RealSense programming unfortunately is not one of my specialist areas.

I do remember though that there have been past discussions about setting FPS and resolution with a json configuration file.  The last time I recall this subject being covered in February 2018, I believe that resolution and FPS could not yet be set outside of the pipeline, as these settings were ignored if they were in a json configuration file.  I do not know if that situation has changed since then.

https://github.com/IntelRealSense/librealsense/issues/1229#issuecomment-368329980

There was a suggestion though about a method used by the RealSense Viewer and the Depth Quality Tool for setting FPS and resolution that seemed to be taking place outside of the pipeline.

https://github.com/IntelRealSense/librealsense/issues/1229?language=en_US#issuecomment-368416538

https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L6692

Dorodnic should be able to help you with this subject.

RealSenseCustomerSupport, 2019-05-03:
----------------------------------------------

Ticket being closed due to inactivity for 30+ days.
Please reopen if further assistance is required. Thank you.

closed 2019-05-10 by RealSenseCustomerSupport