realsenseai/librealsense #3597: Compilation errors compiling under WIndows

claudiofantacci, 2019-03-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | n/a | 
| Firmware Version                   | n/a | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.19.0 }                          | 
| Language                            |  CMake | 
| Segment			|  others                       | 

### Issue Description
In this issue I would like to recap some issues I found so far while trying to compile `librealsense` and make a `vcpkg` port out of it, see https://github.com/Microsoft/vcpkg/pull/5777.

The main issues found so far are related to the automatic generation of the `libtm/fw.h` file (related issue #3483) and getting `libusb` from git (possible related issues that are closed but I think they are unresolved #2968 #2719).

Let's go in order:
 1. `libtm`: apparently, under Windows, writing to a file can be a problem when appending content continuously, as currently done in https://github.com/IntelRealSense/librealsense/blob/ca352a12a0d7e88236007cb7d105e852afbfa61d/third-party/libtm/resources/CMakeLists.txt#L113-L127 (and some other blocks after that). As reported in #3483, the solution found so far is to change those line a single write to file, as in this [patch](https://github.com/Microsoft/vcpkg/pull/5777/files#diff-29ed5aa4a9f20476fe1fd619b13955d4).

This seems to resolve (or mitigate?!) the problem. I'm wondering if it would be better to add a specific SHA check on the generated file, though this would require to hardcode an hash code and maintin it.

 2. `libusb`: this is much more complicated. Getting the code using `ExternalProject_Add` every now and then can fail, more often in test environments like in https://github.com/Microsoft/vcpkg/pull/5777. If this would be the case, there is little we can do even though it would be nice to use a system wide `libusb` that seems to be possible, but not implemented yet, see #3024.
In the meantime, by my understanding, one can _uncheck_ `BUILD_WITH_TM2` to avoid getting `libsub`. This however is triggering two different problems under Windows:
    - If you use `ninja` as build tool, you get a permission error when getting `libusb` because apparently there are multiple processes working in the same directory (`build/third-party`) and the compilation fails. I noted that this is not happening if I use `Visual Studio` because all the external project are first downloaded and then are consumed during compilation. My guess is that `ninja` runs parallel jobs based on build priority, and once `libusb` is required it tries to get it with git triggering a permission error. This behaviour should be reported upstream, but I think there is little we can do as I'm not able to reproduce this in another CMake project. If we want to be more robust to this kind of errors, I suggest to download `libusb`, and any other similar dependencies, in independent folders, and not within a common one like `build/third-party`. I tried to do it and it seems to be working just fine.
    - Having `BUILD_WITH_TM2=OFF` does not skip `libusb` under Windows, even if apparently it should not be required. As suggested [here](https://github.com/IntelRealSense/librealsense/issues/2719#issuecomment-474823042) and as you can see from https://github.com/IntelRealSense/librealsense/blob/ca352a12a0d7e88236007cb7d105e852afbfa61d/CMake/windows_config.cmake#L41, `libusb` is just forced to be downloaded. If you surrond it with a check on `BUILD_WITH_TM2`, the compilation ends succesfully.

To recap, here are the actions I would like to propose:
 1. Update the CMakeLists in `third-party/libtm/resources` to avoid possible header generation errors.
 2. Surround https://github.com/IntelRealSense/librealsense/blob/ca352a12a0d7e88236007cb7d105e852afbfa61d/CMake/windows_config.cmake#L41 with `if(BUILD_WITH_TM2)`.
 3. Move all `ExternalProject_Add` projects to an independent directory, rather than a common one. Note that this only affect the structure of the `build` folder.
 4. **Windows bonus**: can we add `set(CMAKE_DEBUG_POSTFIX "d")` in the `windows_config.cmake` file? This is necessary when an user wants both release and debug library installed in the system and would avoid the user to set it manually 😄 (I know, it's minor, but useful).

bfulkers-i, 2019-03-27:
@claudiofantacci not sure if it will solve your problems but you might take a look at #3507 which has lots of cmake changes including things around libusb.

dorodnic, 2019-03-27:
Thanks @claudiofantacci 
We are making enhancements to CMake and in somewhat longer term (couple of months) planning to get rid of libusb dependency on Windows (and use WinUSB instead)
Perhaps you could make a pull-request with some of the changes you are proposing?

claudiofantacci, 2019-03-27:
> @claudiofantacci not sure if it will solve your problems but you might take a look at #3507 which has lots of cmake changes including things around libusb.

Thanks! I'll test that branch and let you know 👍 

> Thanks @claudiofantacci
> We are making enhancements to CMake and in somewhat longer term (couple of months) planning to get rid of libusb dependency on Windows (and use WinUSB instead)

This is great! Thanks! 👍 

> Perhaps you could make a pull-request with some of the changes you are proposing?

If you are ok with it, I can for sure prepare a PR with all my findings. After that we can discuss what to merge and what to discard 👍

radfordi, 2019-03-28:
@claudiofantacci, thanks for testing #3507.  Glad it worked for you.

I agree that including `libusb` could be protected by `BUILD_WITH_TM2`, but it is also needed by the python bindings, android and `libuvc`.  Better would be for each of the these users to call something like `find_package(usb)`.  That way it's idempotent and only included where it's needed.

`CMAKE_DEBUG_POSTFIX=d` is a user setting/convention and so something you should be passing on the command line (however annoying), not something we should be setting generically.

radfordi, 2019-03-28:
@claudiofantacci, if `file(APPEND` doesn't work on Windows, isn't this a `cmake` bug?  Combining the appends into a single write would be a fine work around for now.

claudiofantacci, 2019-03-29:
Hi @radfordi, thanks for your replies!

> I agree that including `libusb` could be protected by `BUILD_WITH_TM2`, but it is also needed by the python bindings, android and `libuvc`. Better would be for each of the these users to call something like `find_package(usb)`. That way it's idempotent and only included where it's needed.

The surround would only be in the `windows_config.cmake` file and consequently it should only affect Windows build. By my understanding it should be fine. Anyway, #3507 seems to fix this so in my PR #3611 I did not add it.

> `CMAKE_DEBUG_POSTFIX=d` is a user setting/convention and so something you should be passing on the command line (however annoying), not something we should be setting generically.

The `CMAKE_DEBUG_POSTFIX=d` would only be for Windows user, building using a `Visual Studio` (or simulated one) generator/environment. It is very limited to a case were it is actually needed, it won't affect any other build. If this, however, breaks any policy, I will remove it without any problem 😄 

> if `file(APPEND` doesn't work on Windows, isn't this a `cmake` bug? Combining the appends into a single write would be a fine work around for now.

I totally agree with you. It is strange that the `APPEND` is failing in that way. As a matter of the fact, I don't believe that the real problem is in the `APPEND` calls. I'm investigating this and asking experienced people about this. Ultimately, I can check CMake code or upstream to understand whether in some circumstances there is a flush problem on file write or not. What I propose is, honestly, a workaround that seems to work. The commands are equivalent, so they don;t harm the build.

zh-12345, 2019-03-30:
> @claudiofantacci not sure if it will solve your problems but you might take a look at #3507 which has lots of cmake changes including things around libusb.

i also have some problems with compilation with win10 python ,for example "Module 'pyrealsense2' has no 'pointcloud' memberpylint(no-member)"

claudiofantacci, 2019-03-30:
Hi @zh-12345 what problem do you have exactly? What CMake parameter are you using? What `librealsense` version or commit checkaout are you on? Did you try PR #3507?

If you let me know, I can make some tests with your configuration and see what happens.

zh-12345, 2019-03-31:
> Hi @zh-12345 what problem do you have exactly? What CMake parameter are you using? What `librealsense` version or commit checkaout are you on? Did you try PR #3507?
> 
> If you let me know, I can make some tests with your configuration and see what happens.

i am using with D435 on python3.6 /windows ,i just copy the examples code.l find that i can open camera and take pictures ,but the depth picture have no color .and error list the module have nothing.i  what is wrong ?thank you !
[image]
[image]

claudiofantacci, 2019-04-01:
Hey @zh-12345, it seems that you were at least able to compile `librealsense` and hence your issues are unrelated to this issue 😃. You can probably open another issue for your specific question. Anyway, if I have the chance, I'll try to have a look at the Python example and come back to you with some support.

zh-12345, 2019-04-05:
> Hey @zh-12345, it seems that you were at least able to compile `librealsense` and hence your issues are unrelated to this issue 😃. You can probably open another issue for your specific question. [...]

thanks for your reply ! however ,i have opened new issues on [https://github.com/IntelRealSense/librealsense/issues/3630], i am looking forward your help!

PR #3611 (2019-04-29): CMake improvements for Windows build
This PR is a followup of issue #3597 that tries to address the problem of #3483 and to add `CMAKE_DEBUG_POSTFIX` when compiling with `MS Visual Studio` and emulators (those that are recognized by CMake as `MSVC` generators).

The main difference lies in `third-party/libtm/resources/CMakeLists.txt` where multiple `file(APPEND ...)` commands are replaced with a single `file(WRITE ...)` command. This is to resolve a problem where sometimes the files where malformed.
⚠️ Note that strange indentation of the `file(WRITE ...)` commands is necessary to have a prope [... cut]
closed 2019-05-11 by RealSenseCustomerSupport