realsenseai/librealsense #3599: I am trying to use hole filling filter from D415 in ros. I cannot find a declaration of the hole filling filter in the base_realsense_node.cpp . How can I use the hole filling filter in ros

Brolin56, 2019-03-27:


dorodnic, 2019-03-27:
I'm not sure if hole-filling is supported in current ros wrapper. @doronhi ?

Brolin56, 2019-03-28:
@dorodnic I used a [forked]https://github.com/johannesburg/uw-realsensel) library of the Intel realsense library by @Johan Michalove which contained a hole filling filter. 
When I cloned it to a local catkin_ws and did a catkin_make the following error occurs:

> In file included from /usr/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /usr/include/librealsense2/hpp/rs_device.hpp:8,
                 from /usr/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/include/librealsense2/rs.hpp:9,
                 from /home/brolin/catkin_ws/src/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/brolin/catkin_ws/src/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/brolin/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/librealsense2/hpp/rs_frame.hpp: In instantiation of ‘void rs2::frame_callback<T>::on_frame(rs2_frame*) [with T = rs2::asynchronous_syncer; rs2_frame = rs2_frame]’:
/home/brolin/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:1486:1:   required from here
/usr/include/librealsense2/hpp/rs_frame.hpp:1053:13: error: no match for call to ‘(rs2::asynchronous_syncer) (rs2::frame)’
             on_frame_function(frame{ fref });
             ^
realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1188: recipe for target 'realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

As per some of the previous discussion on how to solve this error I tried to upgrade the librealsense2 packages, but I already have the updated files. I have 2.19.1 release of the librealsense2 packages.

Can you help me resolve the issue @dorodnic

doronhi, 2019-03-28:
It is not part of the ros wrapper yet. In the TODO list.

Brolin56, 2019-03-28:
Can you help me with the issue on the forked library @doronhi

johan, 2019-03-28:
@Brolin56 Try to avoid @ signs in front of names that aren't github user names; it notifies unrelated randos.

Brolin56, 2019-03-28:
okay

closed 2019-05-17 by RealSenseCustomerSupport