realsenseai/librealsense #3604: Start Up R200

AngeloEspinoza, 2019-03-27:
Hi there, I'm quite new with the Intel RealSense cameras and I've been struggling with starting up the Intel RealSense R200 for almost 3 days now, I've read all the information necessary and also tested some examples, however I've got some questions about the camera.
First It's worth to say that I could download the DCM and SDK, and opened some of the samples that the SDK comes with and they work fine, however the problem comes when I want to pass to the code, according to my research I cannot use Python to use the library pyrealsense2 since this version (R200) doesn't accept that type of camera, am I wrong? and the second question is about the IDE, which I'm not able to understand where I can start programming or trying my own code with the functionality or components that this camera has, do I have to use C# or C++ necessarily?
The main question is how can I start coding with the camera and using its functionality? since as I mentioned I've got a project to do.

Sincerely,
Angelo D. Espinoza V.

MartyG-RealSense, 2019-03-27:
If you are new to RealSense, I believe that you will find it easier to use the Windows-based '2016 R2' SDK if you have access to Windows and are not using this SDK already.  It can be used with the Unity game creation engine and has pre-made camera scripts that you can drag and drop into objects and configure with menus to enable you to create the basis of a camera-controlled application with a minimum of programming. 
  
'2016 R2' can be downloaded directly in your browser as a 1.8 gb download with the link below:

http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe

If you are using Linux then the 'Legacy Librealsense' SDK, also known as Librealsense 1.12, would be the appropriate SDK to use.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

Instead of Pyrealsense2, the community-supported Pyrealsense can be used with Librealsense 1.12.

https://github.com/toinsson/pyrealsense

In regard to setting up a development environment: if you choose to use the '2016 R2' SDK, the links to the development environment setup information for C++, C#, JavaScript and Java can be found in the side-panel on this page:

https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_devguide_application_development.html

If you choose Librealsense 1.12, one of the Intel team on this forum can provide a better description for correctly setting up a development environment than I can.

AngeloEspinoza, 2019-03-27:
Thanks for the quick answer I really appreciate it, but I think I wasn't that specific and I apologize about it, I already installed the SDK 2016 which was the one that worked fine for the R200, and I want to use this camera together with the library of OpenCV in python with the help of the infrared sensors of the R200 I want to determine distances of objects and so on in realtime.

Right now I'm trying to install the library pyrealsense with the instructions provided by you, however it says something about rs.h which I don't know what it is I know how to set 'new environment variables' however I don't know what the rs.h is about.

I would appreciate you can help with that, if there's missing information about what I want to express, please let me know.

MartyG-RealSense, 2019-03-27:
I have not installed Librealsense 1.12 myself.  When the instructions reefer to the 'rs.h directory location', my guess would be that it means to point to the docs folder of Pyrealsense that contains rs.h

https://github.com/toinsson/pyrealsense/tree/master/docs

AngeloEspinoza, 2019-03-27:
Can you be more specific?

MartyG-RealSense, 2019-03-27:
I have not used 1.12 or Pyrealsense, and unfortunately could not find much further information through research, aside from this:

https://github.com/toinsson/pyrealsense/issues/31

A member of Intel's team or someone in the community who has done this installation will hopefully be able to help with your problem.

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@AngeloEspinoza I understood you're having question of  "Windows specifics: set environment variable PYRS_INCLUDES to the `rs.h` directory location and environment variable PYRS_LIBS to the librealsense binary location. You might also need to have `stdint.h` available in your path.", right?

rs.h should be in the "include/librealsense" in Librealsense 1.12 as you can see in https://github.com/IntelRealSense/librealsense/blob/v1.12.1/include/librealsense/rs.h

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@AngeloEspinoza If no other questions, will close the ticket accordingly.

closed 2019-05-03 by RealSenseCustomerSupport

selamie, 2020-05-30:
> @AngeloEspinoza I understood you're having question of "Windows specifics: set environment variable PYRS_INCLUDES to the `rs.h` directory location and environment variable PYRS_LIBS to the librealsense binary location. You might also need to have `stdint.h` available in your path.", right?
>

Hi, I'm also new to realsense. I've been trying to solve this problem for a while now to install PyRealSense--where is the librealsense binary? What should I be looking for?

Thank you!

MartyG-RealSense, 2020-05-30:
Hi @selamie   Which model of RealSense camera are you using please?

selamie, 2020-05-30:
I'm using the D415 with librealsense2 on Windows 10 and Python 3.7

So I think I got it right (I installed pyrealsense2 successfully and ran some examples) but I'd still like to make sure--currently my PYRS_LIBS variable is pointed at the location of realsense2.lib

Is that file the 'librealsense binary'?

MartyG-RealSense, 2020-05-30:
The use of **PYRS_LIBS** makes me wonder if you are looking at information about **pyrealsense**.  There was an older version of Pyrealsense for cameras such as the RealSense R200, and this Pyrealsense became known as Pyrealsense 2.2 eventually.  This caused people to confuse it with the similarly named but incompatible **Pyrealsense2** that the modern 400 Series cameras use.

May I confirm that you are using **Pyrealsense2** please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#installation