realsenseai/librealsense #3606: Tracking confidence?

antithing, 2019-03-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | t265 | 

| Operating System & Version |   {Win 10  | 

| Platform                                 | PC |
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Is it possible to return the tracking confidence from the t265 camera using the sdk? 
I see the example has liw/medium /high, but can I get a more accurate value? A confidence percentage or similar? 
How trustworthy is the value? 
How is it calculated? 

Thanks!

bfulkers-i, 2019-03-29:
Hi @antithing, the confidence value returned from the T265 is simply low, medium, or high. The value corresponds to the local tracking quality of the device and for most applications you should trust the full 6dof pose only in high confidence. If you only need the rotation (3dof), lower confidence poses can be used.

closed 2019-04-12 by RealSenseCustomerSupport

jattiogbe, 2019-04-17:
Hi, 

Just a couple of questions to be clear on that subject :
we get the value 1, 2 or 3 for tracker_confidence, right ?
1 = low
2 = medium
3 = high
That's it ?

What does the mapper_confidence represent ?

arjunsengupta98, 2020-09-04:
Hello
I would also like to know what mapper confidence signifies.

https://github.com/IntelRealSense/librealsense/issues/6054#issuecomment-687012423