realsenseai/librealsense #3608: I was wondering if there is any video tutorial on configuring the repo to make my f200 3d camera wrk with open ni2 and intel realsense.

Valema1986, 2019-03-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Valema1986, 2019-03-28:
I generally prefer python, but I am new to programming I need a tutorial for learn to do this.

MartyG-RealSense, 2019-03-28:
The closest match I could find to your request was this video:

https://www.youtube.com/watch?v=Bn9WqbYtNBw

Although the title says that it was done on a Jetson Tx1 computing device and a Structure Sensor camera, the comments say that the installation process also works with other types of computing device and depth camera, such as Raspberry Pi board and Kinect camera.

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@MartyG-RealSense Thanks very much for your great support! :)

@Valema1986 Is the tutorial in the video working for you? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@Valema1986 Any other questions for this ticket?

Valema1986, 2019-04-08:
No, not any other question, thanks.

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

@Valema1986 Great! Thanks for the update! Will close the ticket accordingly.

closed 2019-04-16 by RealSenseCustomerSupport

vocong25, 2020-01-17:
@Valema1986  @RealSenseCustomerSupport  have you build librealsense for F200 and Raspberry  Pi successfully ?