realsenseai/librealsense #3612: Odroid XU4 (Android Nougat) freezing after several seconds

daniel-dennis, 2019-03-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |  Android 7.1.2  | 
| Kernel Version (Linux Only)    |  3.10.9 (Ubuntu) | 
| Platform                                 | Odroid XU4 |
| SDK Version                            |  2.0/latest    | 
| Language                            |  Java/C++                   | 
| Segment			| Smartphone          | 

### Issue Description
I've been trying to get an RS camera to work on an Odroid XU4, first by downloading the app from the Google Play store, and also by compiling it myself. The camera feed typically works for a few seconds, but then it freezes. I've tried another XU4, the same issue happens, and the issue doesn't happen on other Android devices with the same camera.

Is there something I should look at in particular to solve this problem?

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@daniel-dennis Which USB port did you use when get the issue? Could you please try if you get the issue when configure only depth or color? Please click the wrench button at the bottom left to configure that. And if you can collect adb logcat logs that will be more helpful for the investigation. Thanks!

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@daniel-dennis Did you see the issue on USB2 or USB3 port? The default configuration of the APP should be depth1280x720@30fps which not supported by USB2. If use USB2, could you please switch to USB3 port to try again?

fubsan, 2019-04-11:
This is reproducible using librealsense 2.19.2 and 2.19.0 on Android 8 and 9 - there seems to be a memory issue - this happens on a high-end Samsung Galaxy Tab S4 10.5 with 6gb ram and USB3.

`
04-09 12:00:31.441 14977 14977 E librs camera streamer: streaming, error: UNKNOWN_ERROR
04-09 12:00:31.614 14977 15011 W librs : Timeout reached waiting for response!
04-09 12:00:31.615 14977 15011 E librs : Read pipe returned error and was clear halted ERROR:Success
04-09 12:00:31.616 712 892 I qti_sensors_hal: processReportInd: Accel - X: -0.171 Y: 0.328 Z: 9.813
04-09 12:00:31.617 14977 15010 W librs : Timeout reached waiting for response!
04-09 12:00:31.619 14977 15010 E librs : Read pipe returned error and was clear halted ERROR:Success
04-09 12:00:31.625 14977 15010 E librs : bad packet: error bit set
04-09 12:00:31.625 14977 15010 E librs : Cannot queue request: Out of memory
`

fubsan, 2019-04-24:
Any news on this? There is clearly a memory leak issue with librealsense on Android.

jonaslindgard, 2019-04-25:
I have tried the com.example.realsense_java_example using librealsense with the .aar file provided from https://dl.bintray.com/intel-realsense/librealsense/com/intel/realsense/librealsense/. 

I'm using a D435 camera with a similar Samsung Galaxy Tab S4 10.5 as mentioned by @fubsan above.

With SDK version 2.19.2, the example app works in that it displays the depth value somewhat reliably, but showing the camera stream to a surface view causes frequent glitches/freezes as described by the original poster.

Today I tested with SDK 2.21.0, and that fails immediately, spamming out of memory traces from librs (both with the unmodified com.example.realsense_java_example and a simple camera preview app based on the official examples):

`2019-04-25 09:56:14.097 16553-16640/com.example.realsense_java_example W/librs: bulk_transfer on read endpoint returned error, ERROR: Out of memory
2019-04-25 09:56:14.098 16553-16640/com.example.realsense_java_example W/librs: bulk_transfer on read endpoint returned error, ERROR: Out of memory
2019-04-25 09:56:14.098 16553-16641/com.example.realsense_java_example W/librs: bulk_transfer on read endpoint returned error, ERROR: Out of memory
2019-04-25 09:56:14.098 16553-16641/com.example.realsense_java_example W/librs: bulk_transfer on read endpoint returned error, ERROR: Out of memory
2019-04-25 09:56:14.098 16553-16640/com.example.realsense_java_example W/librs: bulk_transfer on read endpoint returned error, ERROR: Out of memory
2019-04-25 09:56:14.098 16553-16640/com.example.realsense_java_example W/librs: bulk_transfer on read endpoint returned error, ERROR: Out of memory
`  

The java example app also crashes on app lifecycle changes:

`2019-04-25 10:02:03.692 17314-17314/com.example.realsense_java_example E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.example.realsense_java_example, PID: 17314
    java.lang.RuntimeException: Unable to resume activity {com.example.realsense_java_example/com.example.realsense_java_example.MainActivity}: java.lang.IllegalThreadStateException
        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3805)
        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3845)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1773)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:7002)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)
     Caused by: java.lang.IllegalThreadStateException
        at java.lang.Thread.start(Thread.java:724)
        at com.example.realsense_java_example.MainActivity.onResume(MainActivity.java:37)
        at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1362)
        at android.app.Activity.performResume(Activity.java:7432)`

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

@daniel-dennis Any update from your side? Looking forward to your reply. Thanks!
@jonaslindgard @fubsan I couldn't reproduce the issue using librealsense v2.21 and FW 5.11.6.200 on Samsung Galaxy S8 with Android 8.0.  Could you please provide the full log for the issue? And what's the camera do you use when get the issue? Thanks!

jonaslindgard, 2019-04-29:
I'm using the D435 camera with FW version 05.11.06.200.

[capture_example_logs.zip](https://github.com/IntelRealSense/librealsense/files/3126820/capture_example_logs.zip)

I've attached 2 logcat logs; one using SDK 2.19.2 and the other using SDK 2.21.0.
With 2.19.2, the camera preview works

 OK (with occasional glitches/freezes), but with 2.21.0 it fails immediately.

The app used is the **capture** example (only modification being that librs is used through downloaded .aar binary instead of locally built:

repositories {
    maven {
        url "https://dl.bintray.com/intel-realsense/librealsense"
    }
}

dependencies {
    implementation 'com.intel.realsense:librealsense:2.21.0@aar'
}

daniel-dennis, 2019-04-29:
> @daniel-dennis Any update from your side? Looking forward to your reply. Thanks!
> @jonaslindgard @fubsan I couldn't reproduce the issue using librealsense v2.21 and FW 5.11.6.200 on Samsung Galaxy S8 [...]

When I run the application, the following errors repeatedly appear in Logcat:

> @daniel-dennis Any update from your side? Looking forward to your reply. Thanks!
> @jonaslindgard @fubsan I couldn't reproduce the issue using librealsense v2.21 and FW 5.11.6.200 on Samsung Galaxy S8 [...]

I was getting the same errors as @fubsan. I have managed to get a Realsense camera working properly on the XU4 however using the Realsense library as a backend (packaging into an aar file), and using my own frontend to display the feed (using a Unity project), so I suspect its an issue with the frame buffer not being cleared properly.

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

@jonaslindgard  @daniel-dennis I do see some problem with** ****t**he**capture**example but not the same as yours. The problem I saw here is that the application can start and get streaming normally for the first time but no images any more after kill/stop the application.  While the**_camera_**example works well for me. So could you please try the **_camera_**example first? We will further check the issue of the **_capture_** example.

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@jonaslindgard  @daniel-dennis Did you get any luck with the **_camera_** example? Looking forward to your update. Thanks!

jlindgard, 2019-05-21:
> @jonaslindgard @daniel-dennis Did you get any luck with the **_camera_** example? Looking forward to your update. Thanks!

Which example do you mean? **java_example** (com.example.realsense_java_example)? 
Out of the given example apps (recording, playback, capture), capture seems to be the closest to my use case.

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@jlindgard Sorry for any unclear in previous response. I mean the example tool in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/tools/camera Could you please have a try?

jlindgard, 2019-06-06:
Thank you for the clarification and for looking into this. I tested the example camera tool with com.intel.realsense:librealsense:2.22.0@aar (instead of building native code myself).
For a while it seems to work OK, but disconnecting the camera and reconnecting it seems to cause everything to slow down considerably and ultimately it stops working (app shows just black screen). Some other random behavior also occurs from time to time, e.g. one of the infrared streams stops working (app says 'invalid configuration' and jumps to settings view).

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@jlindgard Thanks for the update! I'll try "disconnect and reconnect" here and update you later.

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@jlindgard @daniel-dennis @jonaslindgard  According to PR#4237 merged to v2.24, the issue will get improved with the PR.  Could you please have a try with v2.24 and share the result? Thanks!

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@jlindgard @daniel-dennis @jonaslindgard Did you get chance to try with v2.24?

RealSenseCustomerSupport, 2019-07-18:
----------------------------------------------

@jlindgard @daniel-dennis @jonaslindgard  If no other questions, will close this ticket.

closed 2019-07-18 by RealSenseCustomerSupport