realsenseai/librealsense #3613: Do I need GPU for using two cameras?

MaryBD, 2019-03-28:
Hello, I am using two Intel RealSense D435, on laptop using python and win 10, Intel i7-6600U CPU 2.6 GHz 2.81 GHz, RAM 16 GB. I apply an obstacle detectection, and color detection algorithms on the cemeras streamings. The streaming is not reliable and every while I loose updating the video streams. Can someone advice me if running two cameras is too much for a laptop and I need GPU.
Thanks.

MartyG-RealSense, 2019-03-28:
The 400 Series cameras contain a hardware component called the Vision Processor D4 that process camera data so that a dedicated GPU on the computer is not required.  Two cameras on your laptop should therefore not be too much.

https://www.stemmer-imaging.com/media/uploads/cameras/13/136603-Intel-RealSense-D4-Vision-Processor-Brochure.pdf

The slowdown may be related to the algorithms.  There was a case in 2018 where the D435 was being used with obstacle avoidance on a drone, and had artifacts appearing on their video image that appeared to them to be severely reducing the performance of their algorithm.

https://forums.intel.com/s/question/0D50P0000490XM0SAM/noise-in-realsense-d435-pointcloud

MaryBD, 2019-03-28:
Thanks for your reply. The algorithm is not that complex and I find out that camera streaming is more reliable when camera in on the tripod rather than when I mount it on the mobile robot (I fix the cable to the camera with screw).

MartyG-RealSense, 2019-03-28:
A week ago you posted a message about reducing vibrations on your robot.  Your description of video loss reminded me of SpaceX space rocket launches where they land the first-stage booster rocket on an ocean barge when it returns to Earth.  Often, the vibration caused by the approaching rocket caused the video stream to cut out, and then it came back as soon as the booster has landed and the vibration has ceased.

Whilst an agricultural robot is not a SpaceX rocket, I wonder if at the times when the image is lost, there might be more vibration being transmitted through the robot's body to the camera mounting than is clear from just looking at it.

MaryBD, 2019-04-01:
Thank you for sharing this. Yes it seems that be the issue, the camera still has lots of vibrations when the robot is moving, I fixed the cable and I thought that might help, by I loose the streamings when robots shakes rapidly and transmits vibrations to the camera. So thanks for the advice I am going to compensate the vibrations even more and see if it helps.

closed 2019-06-14 by RealSenseCustomerSupport