realsenseai/librealsense #3615: Frame control: rewind/seek

soarwing52, 2019-03-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (5.11.1.100(latest)) | 
| Operating System & Version |   {Win10}  | 
| Kernel Version (Linux Only)    |  (N/A)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.19.0(latest) }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Measure }                          | 

### Issue Description
Hi, I'm working on the D35 model, and viewing frames to measure

#3549 I achieved to measure the first frame of a bag, and now I'm working on going to selected frames 
I can go forward with this code here
`while i < 10:
        print i
        for x in range(50):
            pipeline.wait_for_frames()
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        playback.pause()
        # Colorize depth frame to jet colormap
        depth_color_frame = rs.colorizer().colorize(depth_frame)
        # Convert depth_frame to numpy array to render image in opencv
        depth_color_image = np.asanyarray(depth_color_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        cv2.namedWindow("Color Stream", cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Color Stream', color_image)
        cv2.waitKey(1000)
        cv2.destroyAllWindows()
        playback.resume()
        i += 1`

this allows skip 50 frames each time

and if I want to go to selected frame I used this
`frame_number = input('no.? \n')

for x in range(frame_number):
            pipeline.wait_for_frames()
frames = pipeline.wait_for_frames()`

So my question will be:
1.How can I use the playback.seek?
in #1579 says that the seek function is not working, also I tried myself and can't get it to work

2. poll_for_frames is not working
I found in #1887 that the poll_from_frame in python is not working, and I also tried

3.If it's possible to jump back one frame?
Currently my method is stop and restart the pipeline and record the frame as frame_num
and put wait_for_frame in the for loop till frame_num -1
it is kind of slow and much to process, if there is a better way

thank you for all the helps!

soarwing52, 2019-04-24:
self answer:
the error of poll_for_frames is because it can pass NULL value
just add 'if not' statement will fixe the issue
but it didnt improve the matching frame issue, which its the same logic for wait_for_frames, poll_for_frames, and syncer
 so 2 is answered.

for 3 currently is loop to the end and back, takes a bit longer, but seems to be the only way.

seek I cant find the usage yet, but I use a loop to match frame number, through it is taking much time

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

Hi @soarwing52,

Since you clarify and resolve the issue, if not other concern, will close this.

closed 2019-06-03 by soarwing52

soarwing52, 2019-06-03:
@RealSenseCustomerSupport 
actually still one last question, if you can provide an example of using rs.seek() in python?