realsenseai/librealsense #3616: Jetson TX1 RGB frames get lost

abylikhsanov, 2019-03-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       D435i 
| Firmware Version                  Latest
| Operating System & Version |   Ubuntu 16.04   JetPack 3.3   L4T R.28.2.0
| Kernel Version (Linux Only)    |  4.4.48-tegra                                         | 
| Platform                                 |  Jetson TX1
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

When I open realsense-viewer, my depth image seems to be incomplete and when I turn on the RGB, frames get missing and I get slight lag. Also, the poincloud is without RGB image.

I am using JetPack 3.3 and L4T R28.2.0

I have used these scripts to install librealsense:
https://www.jetsonhacks.com/2018/04/25/now-cuda-intel-realsense-d400-cameras-nvidia-jetson-tx/

Do you guys some updated scripts? I am runing v2.19.0

dorodnic, 2019-04-01:
The issue is not with the script (RGB does not require any kernel patches), seems to be with the setup itself. I tested this with both the on-board USB3 and a PCIe USB controller on TX2, in both cases with RGB frame drops. Also, people tested it with [linux built-in tools](https://github.com/IntelRealSense/librealsense/issues/2682) with no-luck. 
Strangely, depth frames of same size do manage to get through...
Something with the Jetson and D400 RGB sensor is not working well, unfortunately. 
Perhaps @RealSenseCustomerSupport could file it for further investigation with the firmware.

RealSenseCustomerSupport, 2019-04-22:
----------------------------------------------

Sorry for the delay but we will take a look and update.
It is also worthwhile to mention that D400 color sensor on a Jetson Xavier over Jetpack 4.2 seems to performs better than on TX2/1.

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

We are still investigating this issue. An internal ticket has also been filed accordingly for further support.

RealSenseCustomerSupport, 2020-03-12:
----------------------------------------------

Is this issue still reproducible after upgrading TX1 to the more recent Jetpack release?
I no longer see the issue on the TX2 with Jetpack 4.2.2.
Thank you.

    nvidia@nvidia-desktop:~$ ./jetsonInfo.py
     Ubuntu 18.04.2 LTS
     Kernel Version: 4.9.140-tegra
     CUDA 10.0.326

    nvidia@nvidia-desktop:~$ rs-enumerate-devices -s
    Device Name                  Serial Number      Firmware Version
    Intel RealSense D415          802312060067        05.12.01.00

    nvidia@nvidia-desktop:~$ ./yavta-master/yavta --capture=30 --size 1280x720 -t 1/30 /dev/video2  -Fimage -fYUYV
    Device /dev/video2 opened: Intel(R) RealSense(TM) Depth Ca (usb-3530000.xhci-1).
    Video format set: width: 1280 height: 720 buffer size: 1843200
    Video format: YUYV (56595559) 1280x720
    Current frame rate: 1/30
    Setting frame rate to: 1/30
    Frame rate set: 1/30
    8 buffers requested.
    length: 1843200 offset: 0
    Buffer 0 mapped at address 0x7fab225000.
    length: 1843200 offset: 1843200
    Buffer 1 mapped at address 0x7fab063000.
    length: 1843200 offset: 3686400
    Buffer 2 mapped at address 0x7faaea1000.
    length: 1843200 offset: 5529600
    Buffer 3 mapped at address 0x7faacdf000.
    length: 1843200 offset: 7372800
    Buffer 4 mapped at address 0x7faab1d000.
    length: 1843200 offset: 9216000
    Buffer 5 mapped at address 0x7faa95b000.
    length: 1843200 offset: 11059200
    Buffer 6 mapped at address 0x7faa799000.
    length: 1843200 offset: 12902400
    Buffer 7 mapped at address 0x7faa5d7000.
    0 (0) [-] 0 1843200 bytes 3828.706252 1584037435.673124
    1 (1) [-] 1 1843200 bytes 3828.739832 1584037435.706773
    2 (2) [-] 2 1843200 bytes 3828.773445 1584037435.739646
    3 (3) [-] 3 1843200 bytes 3828.807029 1584037435.773383
    4 (4) [-] 4 1843200 bytes 3828.840622 1584037435.806482
    5 (5) [-] 5 1843200 bytes 3828.874226 1584037435.840876
    6 (6) [-] 6 1843200 bytes 3828.907819 1584037435.874001
    7 (7) [-] 7 1843200 bytes 3828.941413 1584037435.907846
    8 (0) [-] 8 1843200 bytes 3828.975011 1584037435.941286
    9 (1) [-] 9 1843200 bytes 3829.008613 1584037435.974508
    10 (2) [-] 10 1843200 bytes 3829.042208 1584037436.007979
    11 (3) [-] 11 1843200 bytes 3829.075814 1584037436.041853
    12 (4) [-] 12 1843200 bytes 3829.109393 1584037436.075061
    13 (5) [-] 13 1843200 bytes 3829.142995 1584037436.108943
    14 (6) [-] 14 1843200 bytes 3829.176587 1584037436.142707
    15 (7) [-] 15 1843200 bytes 3829.210192 1584037436.175936
    16 (0) [-] 16 1843200 bytes 3829.243787 1584037436.209937
    17 (1) [-] 17 1843200 bytes 3829.277389 1584037436.243132
    18 (2) [-] 18 1843200 bytes 3829.310983 1584037436.277650
    19 (3) [-] 19 1843200 bytes 3829.344587 1584037436.310686
    20 (4) [-] 20 1843200 bytes 3829.378188 1584037436.344801
    21 (5) [-] 21 1843200 bytes 3829.411780 1584037436.378208
    22 (6) [-] 22 1843200 bytes 3829.445388 1584037436.410977
    23 (7) [-] 23 1843200 bytes 3829.478967 1584037436.445055
    24 (0) [-] 24 1843200 bytes 3829.512578 1584037436.478668
    25 (1) [-] 25 1843200 bytes 3829.546162 1584037436.512271
    26 (2) [-] 26 1843200 bytes 3829.579765 1584037436.545725
    27 (3) [-] 27 1843200 bytes 3829.613369 1584037436.579692
    28 (4) [-] 28 1843200 bytes 3829.646963 1584037436.613300
    29 (5) [-] 29 1843200 bytes 3829.680558 1584037436.646210
    Captured 29 frames in 0.984545 seconds (29.455231 fps, 56164014.849499 B/s).
    8 buffers released.

closed 2020-03-18 by RealSenseCustomerSupport