realsenseai/librealsense #3618: rs-viewer bug

amaanda, 2019-03-28:
When pausing a .bag file on RS Viewer, the 3D vision does pause. But then, when pressing the play button, I have to first press the 3D play button and then the .bag button, because if I just press the general .bag play button the 3D vision won't work. It would be a lot easier if they could be synchronized.

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@amaanda Which librealsense version did you get the issue? And could you please provide a snapshot with the mark of the buttons you mentioned so that we can understand the problem more clearly? Thanks!

amaanda, 2019-04-04:
@RealSenseCustomerSupport I am using 2.19.0.

The red circled button is the general play button, and the blue circled one is the "3D play button".

[image]

RealSenseCustomerSupport, 2019-04-11:
----------------------------------------------

@amaanda I tried v2.20 and can see that the frame for the pause moment was still there after click general play button. This pause frame can be cleared after click "3D play button". Is that the problem what you see?

amaanda, 2019-04-11:
@RealSenseCustomerSupport Yes, it is

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@amaanda Got your point now. We will look into this and update when get fix. Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

Pending on JIRA DSO-12584. No update yet.

aangerma, 2020-11-02:
Hi @amaanda 
the issue fixed on: 
https://github.com/IntelRealSense/librealsense/pull/7700

RealSenseSupport, 2020-11-16:
@amaanda The fix in https://github.com/IntelRealSense/librealsense/pull/7700 now merged to development branch. Could you please have a try? Thanks!

amaanda, 2020-11-16:
Hi @RealSenseSupport I will try it as soon as possible. Thanks! =)

RealSenseSupport, 2020-11-23:
@amaanda SDK v2.40 includes the fix for the issue and released now. Please try it with RealSenseViewer. Thanks!

RealSenseSupport, 2020-12-01:
@amaanda As the issue resolve, closed the issue accordingly. Thanks!

closed 2020-12-01 by RealSenseSupport