realsenseai/librealsense #3619: Viewer Depth Visualization Parameters

JBBee, 2019-03-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  2.19.2                          | 
| Language                            |  Unity                         | 
| Segment			|  AR                         | 

### Issue Description
Question about the 'Depth Visualization' controls in the Realsense Viewer: Do these control parameters to the internal SLAM algorithm running on the T265?  If not what do these parameters control, for example the min distance and max distance sliders?

Thanks.

MartyG-RealSense, 2019-03-28:
Section 3.3 on page 11 onward of the Viewer user guide PDF explains the Depth Visualization functions.

https://www.intel.co.uk/content/www/uk/en/support/articles/000028593/emerging-technologies/intel-realsense-technology.html

JBBee, 2019-03-28:
Thanks @MartyG-RealSense but that document was last revised in May of 2018, before the release of the T265 so there's no reference to the T265 in there.  Those depth visualization options are visible when the Tracking Module is plugged in, so I'm wondering what (if anything) they control.  There's no depth image available from the T265.

MartyG-RealSense, 2019-03-28:
The depth visualization options have been around since the launch of the RealSense Viewer, pre-dating the T265 by a long time.  I do not have a T265 so I do not have information on why the depth visualization options are present when only the T265 is plugged in, unless it is for when a depth camera such as the D435 is plugged in alongside the T265 (such a camera can be paired with the T265 to capture depth data).

bfulkers-i, 2019-03-29:
@JBBee the depth visualization options exposed in Realsense Viewer do not control anything on the T265, it's a minor known issue in the visualization that causes them to be displayed (cc @dorodnic)

closed 2019-04-01 by JBBee