realsenseai/librealsense #3620: How to make arr file?

GameHuiTea, 2019-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.19.1 }                          | 
| Language                            |  { unity }                          | 
| Segment			|  { Smartphone }                          | 

### Issue Description
<question>
I developed in a Unity. And I want to build it as an apk and make it work on Android.
In order to do that, we had to create a .arr file to replace realsense2.dll.
I tried to follow this link in both ways, but I failed.
https://github.com/IntelRealSense/librealsense/blob/development/wrappers/android/readme.md

**Below are the failed logs from Android Studio.**

오후 12:16:02: Executing task 'assembleRelease'...

> Configure project :camera
Observed package id 'tools' in inconsistent location 'C:\Users\TeamElysium\AppData\Local\Android\Sdk\tools-X' (Expected 'C:\Users\TeamElysium\AppData\Local\Android\Sdk\tools')
Already observed package id 'tools' in 'C:\Users\TeamElysium\AppData\Local\Android\Sdk\tools'. Skipping duplicate at 'C:\Users\TeamElysium\AppData\Local\Android\Sdk\tools-X'

> Task :camera:checkReleaseClasspath
> Task :camera:preBuild UP-TO-DATE
> Task :librealsense:preBuild UP-TO-DATE
> Task :librealsense:preReleaseBuild UP-TO-DATE
> Task :librealsense:checkReleaseManifest
> Task :librealsense:processReleaseManifest
> Task :camera:preReleaseBuild
> Task :librealsense:compileReleaseAidl NO-SOURCE
> Task :camera:compileReleaseAidl NO-SOURCE
> Task :librealsense:packageReleaseRenderscript NO-SOURCE
> Task :camera:compileReleaseRenderscript
> Task :camera:checkReleaseManifest
> Task :camera:generateReleaseBuildConfig
> Task :camera:prepareLintJar UP-TO-DATE
> Task :camera:mainApkListPersistenceRelease
> Task :camera:generateReleaseResValues
> Task :camera:generateReleaseResources
> Task :librealsense:compileReleaseRenderscript
> Task :librealsense:generateReleaseResValues
> Task :librealsense:generateReleaseResources
> Task :librealsense:packageReleaseResources
> Task :camera:mergeReleaseResources
> Task :camera:createReleaseCompatibleScreenManifests
> Task :camera:processReleaseManifest
> Task :camera:splitsDiscoveryTaskRelease
> Task :librealsense:generateReleaseRFile
> Task :camera:processReleaseResources
> Task :camera:generateReleaseSources
> Task :librealsense:generateReleaseBuildConfig
> Task :librealsense:prepareLintJar UP-TO-DATE
> Task :librealsense:generateReleaseSources
> Task :librealsense:javaPreCompileRelease

> Task :librealsense:compileReleaseJavaWithJavac
Note: Some input files use unchecked or unsafe operations.
[... 177 lines cut ...]
  [3/111] Building CXX object ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-options.cpp.o
  FAILED: ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-options.cpp.o
  ccache C:\Users\TeamElysium\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=x86_64-none-linux-android21 --gcc-toolchain=C:/Users/TeamElysium/AppData/Local/Android/Sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/TeamElysium/AppData/Local/Android/Sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/sysroot  -DBUILD_EASYLOGGINGPP -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DRS2_USE_ANDROID_BACKEND -DUNICODE -Drealsense2_EXPORTS -I../../../../../../../realsense2 -I../../../../../../../src -I../../../../../../../third-party/realsense-file/rosbag/console_bridge/include -I../../../../../../../third-party/realsense-file/rosbag/cpp_common/include -I../../../../../../../third-party/realsense-file/rosbag/rosbag_storage/include -I../../../../../../../third-party/realsense-file/rosbag/roscpp_serialization/include -I../../../../../../../third-party/realsense-file/rosbag/rostime/include -I../../../../../../../third-party/realsense-file/rosbag/roscpp_traits/include -I../../../../../../../third-party/realsense-file/rosbag/roslz4/include -I../../../../../../../third-party/realsense-file/rosbag/msgs -I../../../../../../../third-party/realsense-file/boost -I../../../../../../../third-party/realsense-file/lz4/lib -I../../../../../../../include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -fno-addrsig -Wa,--noexecstack -Wformat -Werror=format-security -stdlib=libc++  -fPIC -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -O2 -DNDEBUG  -fPIC   -std=gnu++11 -MD -MT ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-options.cpp.o -MF ..\..\..\..\..\build\CMakeFiles\realsense2.dir\src\ds5\ds5-options.cpp.o.d -o ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-options.cpp.o -c C:/Realsense2191/src/ds5/ds5-options.cpp
  ccache: FATAL: Could not find compiler "C:\Users\TeamElysium\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe" in PATH
  [4/111] Building CXX object ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-rolling-shutter.cpp.o
  FAILED: ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-rolling-shutter.cpp.o
  ccache C:\Users\TeamElysium\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=x86_64-none-linux-android21 --gcc-toolchain=C:/Users/TeamElysium/AppData/Local/Android/Sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/TeamElysium/AppData/Local/Android/Sdk/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/sysroot  -DBUILD_EASYLOGGINGPP -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DRS2_USE_ANDROID_BACKEND -DUNICODE -Drealsense2_EXPORTS -I../../../../../../../realsense2 -I../../../../../../../src -I../../../../../../../third-party/realsense-file/rosbag/console_bridge/include -I../../../../../../../third-party/realsense-file/rosbag/cpp_common/include -I../../../../../../../third-party/realsense-file/rosbag/rosbag_storage/include -I../../../../../../../third-party/realsense-file/rosbag/roscpp_serialization/include -I../../../../../../../third-party/realsense-file/rosbag/rostime/include -I../../../../../../../third-party/realsense-file/rosbag/roscpp_traits/include -I../../../../../../../third-party/realsense-file/rosbag/roslz4/include -I../../../../../../../third-party/realsense-file/rosbag/msgs -I../../../../../../../third-party/realsense-file/boost -I../../../../../../../third-party/realsense-file/lz4/lib -I../../../../../../../include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -fno-addrsig -Wa,--noexecstack -Wformat -Werror=format-security -stdlib=libc++  -fPIC -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -O2 -DNDEBUG  -fPIC   -std=gnu++11 -MD -MT ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-rolling-shutter.cpp.o -MF ..\..\..\..\..\build\CMakeFiles\realsense2.dir\src\ds5\ds5-rolling-shutter.cpp.o.d -o ../../../../../build/CMakeFiles/realsense2.dir/src/ds5/ds5-rolling-shutter.cpp.o -c C:/Realsense2191/src/ds5/ds5-rolling-shutter.cpp
  ccache: FATAL: Could not find compiler "C:\Users\TeamElysium\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe" in PATH
[... cut ...] c/ds5/ds5-timestamp.cpp
  ccache: FATAL: Could not find compiler "C:\Users\TeamElysium\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe" in PATH
  ninja: build stopped: subcommand failed.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 13s
39 actionable tasks: 2 executed, 37 up-to-date

GameHuiTea, 2019-04-02:
Is there a problem with my question? If there's a problem, I'll fix it. I need you to answer anyone.

dorodnic, 2019-04-02:
Hi @GameHuiTea 
Sorry, no problem with the question at all, we are just temporarily overwhelmed on all projects.
We make sure all issues get addressed eventually, and I asked @matkatz to comment. 
Sorry for the delay

matkatz, 2019-04-10:
Hi, @GameHuiTea 
Sorry for the late response, it looks like a setup issue, are you able to build any NDK project on your system?
In case you want to skip building the library, you can download the AAR from [here ](https://dl.bintray.com/intel-realsense/librealsense/com/intel/realsense/librealsense/2.19.2/)

GameHuiTea, 2019-04-11:
Hi, @matkatz 
Thank a lot for reply.

I tried to run unity-built android app in my chromebook(android 7.1.1). But still have some problem that my device can't recognize a realsense.

To fix this problem, I will do two things. I didn't root my android device yet, so I will try to run again after rooting my device. And as your advice, I will also check whether I can build other NDK project (sample project maybe?).

However, could you tell me about your environment relative to NDK project setting? That may be helpful to me. And one more thing, I got an error that there is no ".gradle" directory when building with gradle. Is this relevant to my setup issue?

matkatz, 2019-04-19:
I'm working with standard Android studio installation, NDK version is 19.1.5304403.
And yes, the gradle directory is relevant for building librealsnese so and AAR.
Did you try to work with the AAR that I have provided?

GameHuiTea, 2019-04-19:
Yes, I tried realsense version 2.19.1 source code to build ARR file.
[image]

But i meet error that there is no ".gradle" directory.
My NDK version is 19.2.5345600.
And my Android studio version is 3.32 (build 182.5107.16.33.5314842)

I'll change the ndk version, try it, and let you know the results. Thank you for your answer!

GameHuiTea, 2019-04-22:
sorry. not ".gradle".
Please check this error.
[image]

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

the error in your screenshot is gradle/android studio issue.
the file "gradle/wrapper/gradle-wrapper.properties" isn't exist in librealsense package.

I think you can click the "OK" button.

GameHuiTea, 2019-04-29:
I don't think I understand the this problem. If I'll find the problem, I reopen the issue.
Thank you for answering my questions!

closed 2019-04-29 by GameHuiTea