realsenseai/librealsense #3621: Is anyone else can offer the precise intrinsic and extrinsic parameter  of T265?

zhuoyueljl, 2019-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

bfulkers-i, 2019-03-29:
Hi @zhuoyueljl, T265 intrinsics and extrinsics will be exposed soon in a future librealsense release. You can follow the progress on #3563 and #3541

dorodnic, 2019-04-07:
Please try with 2.20
(you can either run `rs-enumerate-devices -c` for raw data or `rs-ar-basic` for a demo)

zhuoyueljl, 2019-04-08:
Hi @dorodnic  I have tried with 2.20 and run  ` rs-enumerate-devices -c  ` , but it return: `Command DEV_GET_CAMERA_INTRINSICS is not supported by FW`

zhuoyueljl, 2019-04-10:
Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format   
    Fisheye 1	  848x800	@ 30Hz	   Y8
    Fisheye 2	  848x800	@ 30Hz	   Y8
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 62Hz	   MOTION_XYZ32F
    Pose	 N/A		@ 200Hz	   6DOF

Provided Intrinsic:
 Intrinsic of "Fisheye 1" / 848x800 / {Y8}
  Width:      	848
  Height:     	800
  PPX:        	419.467010498047
  PPY:        	386.97509765625
  Fx:         	286.221588134766
  Fy:         	287.480102539062
  Distortion: 	Kannala Brandt4
  Coeffs:     	-0.0043481751345098  	0.037125650793314  	-0.0355393998324871  	0.00577297387644649  	0  

 Intrinsic of "Fisheye 2" / 848x800 / {Y8}
  Width:      	848
  Height:     	800
  PPX:        	422.039001464844
  PPY:        	397.015991210938
  Fx:         	285.229187011719
  Fy:         	286.333709716797
  Distortion: 	Kannala Brandt4
  Coeffs:     	-0.00527892587706447  	0.0386337004601955  	-0.0366357117891312  	0.00610797293484211  	0  

Provided Motion Intrinsic:
Motion Intrinsic of "Gyro"	  MOTION_XYZ32F
Bias Variances: 4.99999998737621e-07  4.99999998737621e-07  4.99999998737621e-07  
Noise Variances: 5.14803014084464e-06  5.14803014084464e-06  5.14803014084464e-06  
Sensitivity : 
     0.996874              0              0     0.00085474  
            0        1.00038              0    -0.00158726  
            0              0       0.992169     0.00233547  

Motion Intrinsic of "Accel"	  MOTION_XYZ32F
Bias Variances: 9.99999974737875e-05  9.99999974737875e-05  9.99999974737875e-05  
Noise Variances: 6.69524524710141e-05  6.69524524710141e-05  6.69524524710141e-05  
Sensitivity : 
      1.01468              0              0      -0.162711  
            0        1.02097              0       0.250464  
            0              0        1.01948       0.114724  

Provided Extrinsic:
Extrinsic from "Fisheye 1"	  To	  "Fisheye 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Fisheye 1"	  To	  "Fisheye 2" :
 Rotation Matrix:
   0.999972        -0.00403926       0.00633063    
   0.00400718       0.999979         0.00507102    
  -0.00635099      -0.00504551       0.999967      

 Translation Vector: -0.0643017739057541  -0.00040316634112969  0.00050472462316975  

Extrinsic from "Fisheye 1"	  To	  "Gyro" :
 Rotation Matrix:
  -0.999935         0.00326365       0.0109447     
  -0.0032628       -0.999995         9.63121e-05   
   0.010945         6.05954e-05      0.99994       

 Translation Vector: -0.0536035150289536  -0.000310176896164194  0.000351920520188287  

Extrinsic from "Fisheye 1"	  To	  "Accel" :
 Rotation Matrix:
  -0.999935         0.00326365       0.0109447     
  -0.0032628       -0.999995         9.63121e-05   
   0.010945         6.05954e-05      0.99994       

 Translation Vector: -0.0536035150289536  -0.000310176896164194  0.000351920520188287  

Extrinsic from "Fisheye 2"	  To	  "Fisheye 1" :
 Rotation Matrix:
   0.999972         0.00400718      -0.00635099    
  -0.00403926       0.999979        -0.00504551    
   0.00633063       0.00507102       0.999967      

 Translation Vector: 0.0643047839403152  0.000145973026519641  -9.55924042500556e-05  

Extrinsic from "Fisheye 2"	  To	  "Fisheye 2" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Fisheye 2"	  To	  "Gyro" :
 Rotation Matrix:
  -0.999989        -0.000743985      0.00459421    
   0.000721044     -0.999987        -0.0049931     
   0.00459787      -0.00498973       0.999977      

 Translation Vector: 0.0106993019580841  5.05454445374198e-05  -8.46007169457152e-05  

Extrinsic from "Fisheye 2"	  To	  "Accel" :
 Rotation Matrix:
  -0.999989        -0.000743985      0.00459421    
   0.000721044     -0.999987        -0.0049931     
   0.00459787      -0.00498973       0.999977      

 Translation Vector: 0.0106993019580841  5.05454445374198e-05  -8.46007169457152e-05  

Extrinsic from "Gyro"	  To	  "Fisheye 1" :
 Rotation Matrix:
  -0.999935        -0.0032628        0.010945      
   0.00326365      -0.999995         6.05954e-05   
   0.0109447        9.63121e-05      0.99994       

 Translation Vector: -0.0536048784852028  -0.000135253212647513  0.000234807026572526  

Extrinsic from "Gyro"	  To	  "Fisheye 2" :
 Rotation Matrix:
  -0.999989         0.000721044      0.00459787    
  -0.000743985     -0.999987        -0.00498973    
   0.00459421      -0.0049931        0.999977      

 Translation Vector: 0.0106995403766632  5.80827982048504e-05  3.56962555088103e-05  

Extrinsic from "Gyro"	  To	  "Gyro" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Gyro"	  To	  "Accel" :
 Rotation Matrix:
   1  -0   0
  -0   1  -0
   0  -0   1

 Translation Vector: 0  0  0  

Extrinsic from "Accel"	  To	  "Fisheye 1" :
 Rotation Matrix:
  -0.999935        -0.0032628        0.010945      
   0.00326365      -0.999995         6.05954e-05   
   0.0109447        9.63121e-05      0.99994       

 Translation Vector: -0.0536048784852028  -0.000135253212647513  0.000234807026572526  

Extrinsic from "Accel"	  To	  "Fisheye 2" :
 Rotation Matrix:
  -0.999989         0.000721044      0.00459787    
  -0.000743985     -0.999987        -0.00498973    
   0.00459421      -0.0049931        0.999977      

 Translation Vector: 0.0106995403766632  5.80827982048504e-05  3.56962555088103e-05  

Extrinsic from "Accel"	  To	  "Gyro" :
 Rotation Matrix:
   1  -0   0
  -0   1  -0
   0  -0   1

 Translation Vector: 0  0  0  

Extrinsic from "Accel"	  To	  "Accel" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0

zhuoyueljl, 2019-04-10:
`Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format   
    Fisheye 1	  848x800	@ 30Hz	   Y8
    Fisheye 2	  848x800	@ 30Hz	   Y8
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 62Hz	   MOTION_XYZ32F
    Pose	 N/A		@ 200Hz	   6DOF

Provided Intrinsic:
 Intrinsic of "Fisheye 1" / 848x800 / {Y8}
  Width:      	848
  Height:     	800
  PPX:        	419.467010498047
  PPY:        	386.97509765625
  Fx:         	286.221588134766
  Fy:         	287.480102539062
  Distortion: 	Kannala Brandt4
  Coeffs:     	-0.0043481751345098  	0.037125650793314  	-0.0355393998324871  	0.00577297387644649  	0  

 Intrinsic of "Fisheye 2" / 848x800 / {Y8}
  Width:      	848
  Height:     	800
  PPX:        	422.039001464844
  PPY:        	397.015991210938
  Fx:         	285.229187011719
  Fy:         	286.333709716797
  Distortion: 	Kannala Brandt4
  Coeffs:     	-0.00527892587706447  	0.0386337004601955  	-0.0366357117891312  	0.00610797293484211  	0  

Provided Motion Intrinsic:
Motion Intrinsic of "Gyro"	  MOTION_XYZ32F
Bias Variances: 4.99999998737621e-07  4.99999998737621e-07  4.99999998737621e-07  
Noise Variances: 5.14803014084464e-06  5.14803014084464e-06  5.14803014084464e-06  
Sensitivity : 
     0.996874              0              0     0.00085474  
            0        1.00038              0    -0.00158726  
            0              0       0.992169     0.00233547  

Motion Intrinsic of "Accel"	  MOTION_XYZ32F
Bias Variances: 9.99999974737875e-05  9.99999974737875e-05  9.99999974737875e-05  
Noise Variances: 6.69524524710141e-05  6.69524524710141e-05  6.69524524710141e-05  
Sensitivity : 
      1.01468              0              0      -0.162711  
            0        1.02097              0       0.250464  
            0              0        1.01948       0.114724  

Provided Extrinsic:
Extrinsic from "Fisheye 1"	  To	  "Fisheye 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Fisheye 1"	  To	  "Fisheye 2" :
 Rotation Matrix:
   0.999972        -0.00403926       0.00633063    
   0.00400718       0.999979         0.00507102    
  -0.00635099      -0.00504551       0.999967      

 Translation Vector: -0.0643017739057541  -0.00040316634112969  0.00050472462316975  

Extrinsic from "Fisheye 1"	  To	  "Gyro" :
 Rotation Matrix:
  -0.999935         0.00326365       0.0109447     
  -0.0032628       -0.999995         9.63121e-05   
   0.010945         6.05954e-05      0.99994       

 Translation Vector: -0.0536035150289536  -0.000310176896164194  0.000351920520188287  

Extrinsic from "Fisheye 1"	  To	  "Accel" :
 Rotation Matrix:
  -0.999935         0.00326365       0.0109447     
  -0.0032628       -0.999995         9.63121e-05   
   0.010945         6.05954e-05      0.99994       

 Translation Vector: -0.0536035150289536  -0.000310176896164194  0.000351920520188287  

Extrinsic from "Fisheye 2"	  To	  "Fisheye 1" :
 Rotation Matrix:
   0.999972         0.00400718      -0.00635099    
  -0.00403926       0.999979        -0.00504551    
   0.00633063       0.00507102       0.999967      

 Translation Vector: 0.0643047839403152  0.000145973026519641  -9.55924042500556e-05  

Extrinsic from "Fisheye 2"	  To	  "Fisheye 2" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Fisheye 2"	  To	  "Gyro" :
 Rotation Matrix:
  -0.999989        -0.000743985      0.00459421    
   0.000721044     -0.999987        -0.0049931     
   0.00459787      -0.00498973       0.999977      

 Translation Vector: 0.0106993019580841  5.05454445374198e-05  -8.46007169457152e-05  

Extrinsic from "Fisheye 2"	  To	  "Accel" :
 Rotation Matrix:
  -0.999989        -0.000743985      0.00459421    
   0.000721044     -0.999987        -0.0049931     
   0.00459787      -0.00498973       0.999977      

 Translation Vector: 0.0106993019580841  5.05454445374198e-05  -8.46007169457152e-05  

Extrinsic from "Gyro"	  To	  "Fisheye 1" :
 Rotation Matrix:
  -0.999935        -0.0032628        0.010945      
   0.00326365      -0.999995         6.05954e-05   
   0.0109447        9.63121e-05      0.99994       

 Translation Vector: -0.0536048784852028  -0.000135253212647513  0.000234807026572526  

Extrinsic from "Gyro"	  To	  "Fisheye 2" :
 Rotation Matrix:
  -0.999989         0.000721044      0.00459787    
  -0.000743985     -0.999987        -0.00498973    
   0.00459421      -0.0049931        0.999977      

 Translation Vector: 0.0106995403766632  5.80827982048504e-05  3.56962555088103e-05  

Extrinsic from "Gyro"	  To	  "Gyro" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Gyro"	  To	  "Accel" :
 Rotation Matrix:
   1  -0   0
  -0   1  -0
   0  -0   1

 Translation Vector: 0  0  0  

Extrinsic from "Accel"	  To	  "Fisheye 1" :
 Rotation Matrix:
  -0.999935        -0.0032628        0.010945      
   0.00326365      -0.999995         6.05954e-05   
   0.0109447        9.63121e-05      0.99994       

 Translation Vector: -0.0536048784852028  -0.000135253212647513  0.000234807026572526  

Extrinsic from "Accel"	  To	  "Fisheye 2" :
 Rotation Matrix:
  -0.999989         0.000721044      0.00459787    
  -0.000743985     -0.999987        -0.00498973    
   0.00459421      -0.0049931        0.999977      

 Translation Vector: 0.0106995403766632  5.80827982048504e-05  3.56962555088103e-05  

Extrinsic from "Accel"	  To	  "Gyro" :
 Rotation Matrix:
   1  -0   0
  -0   1  -0
   0  -0   1

 Translation Vector: 0  0  0  

Extrinsic from "Accel"	  To	  "Accel" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0 `

bfulkers-i, 2019-04-10:
Hi @zhuoyueljl it looks like you were able to use librealsense `2.20.0` to retrieve the intrinsics and extrinsics, can I close this issue?

closed 2019-04-12 by RealSenseCustomerSupport