realsenseai/librealsense #3622: macOS Installation still have "ld: library not found for -lusb-1.0"

Core055, 2019-03-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   {MacOS}  | 
| SDK Version                            |  { legacy / 2.<16>.<5> }                          | 
| Language                            |  {C++}                          | 

when I do as 
"/bin/launchctl setenv LIBRARY_PATH /usr/local/lib"  ,  there still have  the following question
[image]

,is the LIBRARY_PATH wrong?

I do it in iterm2 , to create a application using Qt.
But now ,the first step is wrong.

Thank you please!

Core055, 2019-03-29:
[image]

closed 2019-03-30 by Core055

Tenkir, 2019-04-02:
@Core055 Did you find a solution?

Core055, 2019-04-03:
Yes! I’ve solved it