realsenseai/librealsense #3623: unpack_yuy2 cuda mode

wonshik, 2019-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (05.10.13.00) | 
| Operating System & Version |   Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.4.38-tegra)                                         | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  { 2.17.1 }                          | 
| Language                            |     C/C++     | 
| Segment			|  {Robot}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Even though I enable the `BUILD_WITH_CUDA`, I can see `unpack_yuy2` process in profiler.
So, I check 

> image.cpp +345

There is `#ifdef` for cuda acceleration. but the problem is that after cuda calculation, CPU unpack code execute again.
I think CPU unpack code should be blocked when `BUILD_WITH_CUDA` is enabled.

Please check this issue.
Thanks

dorodnic, 2019-03-29:
Hi @wonshik 
It is not entirely unlikely that this was broken in one of the merges. At the moment, we do not **yet officially** support or validate RealSense on Jetson (though working in that direction). 
I'll look into it.  
If you have a patch for this issue, we accept and encourage [community contribution](https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md)

dorodnic, 2019-04-01:
Confirmed, pull-request submitted. 
Thank you for pointing this out.

dorodnic, 2019-04-07:
Should be resolved in 2.20

closed 2019-04-07 by dorodnic