realsenseai/librealsense #3624: Depth units not taken into account with rosbag and realsense-viewer

s-trinh, 2019-03-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Ubuntu 16.04  | 
| SDK Version                            | 2.19.1                          | 

### Issue Description
With `realsense-viewer`, it is possible to change the depth units in `Advanced Controls > Depth Table` to be able to "see farther". When recording a rosbag and then loading it, the depth units is correctly loaded but it looks like the visualization or the saved pointcloud does not take into account the depth units.

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

@s-trinh Could you please provide some snapshots of the issue so that we can understand it more clearly? Thanks!

s-trinh, 2019-04-30:
Following screenshot is with default depth unit = 1000:
[image]

Increasing the depth unit allows "seeing farther" (see also this [BKM presentation](https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf):

[image]

Setting the depth unit then recording a rosbag and loading it, the depth unit set is not taken into account for the visualization:

[image]

It would be great to either save the depth unit in the rosbag or be able to change it in the viewer when loading a rosbag.

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@s-trinh Please note that the depth unit is not the major fact to see further distance when the target is less than 6.5m. seen from your snapshot of "Depth Unit", the setting should have been configured already. The reason that why you can't see further distance when playback is due to there's threshold filter in post-processing. Please disable "Post-Processing -> Threshold Filter" or adjust "Max Distance" to a bigger value, then you can see further.

s-trinh, 2019-05-06:
Thank you, the "Threshold Filter" parameter was the issue.

closed 2019-05-06 by s-trinh