realsenseai/librealsense #3625: Tune depth stream for outdoor usage, D415

TDVL, 2019-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

TDVL, 2019-03-29:
Also, I have a question about settings. I would like to use the RealSense D415 outdoors with high accuracy setting? Are there any documents about these settings or a preset JSON file? Where can I find the documents for stereo control, advanced control, depth visualisation and postprocessing settings? I need the explanation of all parameters in those setting in order to try to tune them. I can't find any documents about this. Or, If there is a document for tuning the setting for outdoor usage, I will be thankful if someone would provide me it.

Thank you in advance.

Kind regards

MartyG-RealSense, 2019-03-29:
Intel has an illustrated camera tuning guide, which includes tuning for use outdoors in sunlight.

https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf?language=en_US

The link below, meanwhile, contains details of Visual Presets, with links to JSON files for different pre-made configurations.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

A guide to post-processing filters can be found here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md

And Section 3.3 on page 11 onward of the RealSense Viewer user guide PDF explains the Depth Visualization functions.

https://www.intel.co.uk/content/www/uk/en/support/articles/000028593/emerging-technologies/intel-realsense-technology.html

TDVL, 2019-03-29:
Hi Marty,
Thank you. I have found most of the settings for edge processing, good visualization indoor, post processing indoor, but I don't have any documents for outdoor usage. I saw the presentation and I have read the explanation about sunlight, but I need some suggestions for parameters settings.
Best regards

MartyG-RealSense, 2019-03-29:
The easiest thing to do would be to have auto-exposure enabled and let the camera's Vision Processor D4 hardware component automatically handle the adjustments depending on current outdoor environmental conditions.

If you prefer to use manual settings outdoors, the Chief Technical Officer of the RealSense Group has some excellent advice in this link:

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446788177

closed 2019-05-30 by RealSenseCustomerSupport

kzobov, 2020-03-04:
> Intel has an illustrated camera tuning guide, which includes tuning for use outdoors in sunlight. [...]

New link for the camera tuning guide is:
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

Webinar replay just in case: https://software.seek.intel.com/RealSense_BKMWebinar-reg

File from the Webinar is here: https://www.intelrealsense.com/wp-content/uploads/2020/03/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf