realsenseai/librealsense #3626: No camera connected

manhha1402, 2019-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300 } | 
| Firmware Version                   | (Open RealSense Viewer --> v2.18.1) | 
| Operating System & Version |   {Linux (Ubuntu 16.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0-46-generic)                                         | 
| Platform                                 | PC  |
|                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I had a new PC and installed the librealsense library successfully, but there is no camera connected when I connect sr300 camera, using "rs-multicam".
The attached file is output of lsusb and dmesg when camera is plugged in:
[image]

MartyG-RealSense, 2019-03-29:
There was a case recently that had the exact same set of errors as the ones at the bottom of your log.  This may be a useful starting point for you.

https://github.com/IntelRealSense/librealsense/issues/3426

ev-mp, 2019-04-17:
@manhha1402 hi, do you still need assistance with this ?

According to the attached screenshot SR300 is not identified in Ubuntu. The required VID:PID in `lsusb` should be `8086:0aa5`.
The two most common factors for SR300 are lack of sufficient power on USB3 port, and/or device being recognized as USB2.

manhha1402, 2019-04-17:
I think the reason is from my laptop. I will check it. Thanks !

closed 2019-04-17 by manhha1402