realsenseai/librealsense #3627: Repeated colour frames when streaming high res colour and depth

Murded, 2019-03-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  Win 10  |                                 | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.2.668 | 
| Language                            | C# | 

### Issue Description
After copying the example at 
[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-2-capture/Window.xaml.cs](url)

I've noticed that changing the colour stream resolution to 1920 x 1080 and the depth resolution to 1280 x 720 the pipeline will continue to return the same colour frame multiple times before a new one is returned.  Example table below (Not using accurate timestamps):

|Colour Timestamp                     |                        Depth Timestamp                                        |
|---------------------------------|------------------------------------------- |
| 1                     |  1
|1                   | 2 | 
| 1 |  3  |                                 | 
| 1 |4  |
| 2 | 5 | 
| 2 | 5 | 

I've just checked and this also happens on the realsesnse viewer so I fear that it could perhaps be a limitation with USB?

Any suggestions or explanations on this issue would be much appreciate

dorodnic, 2019-04-01:
This will happen when some of the frames are getting dropped at USB / driver level. 
(Instead of returning an error from `get_depth_frame` or `get_color_frame` and crashing the app, we just provide the latest available frame of that time) 
As for the source of the problem, many things can cause frame drops, but one thing you should check is that you are building the SDK as Release (assuming you build it from source)

Murded, 2019-04-01:
Hi @dorodnic thanks for the response,  I thougth this could be a USB issue. I didn't build from source, I grabbed the DLLS from the latest RealsenseViewer so I'm not sure if these are built for release or not.  I'll continue to compare the devices that it works and does not work on in particular the USB ports.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@Murded Any update and any other questions about the ticket? Looking forward to your update. Thanks!

Murded, 2019-04-26:
> @Murded Any update and any other questions about the ticket? Looking forward to your update. Thanks!

I haven't had time to compare devices although I'm satisfied that's likely a USB and or Driver issue so feel free to close the issue

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

@Murded Thanks for the update! Close this ticket now. Please feel free to open new if you have any other questions or issues.

closed 2019-05-02 by Murded