realsenseai/librealsense #3628: API access to the feature key points used in tracking from T265

dperox, 2019-03-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Feature request - API access to the feature key points used in tracking from T265:
- Can be either as a per frame matched features or stored key frame features 
- The features should contain a 3D position of each feature in world coordinates
- Add a callback whenever new features are added to the map

The main premise behind this FR is to get a sparse depth map

leo199483, 2019-04-22:
I am also looking for a function that can extract feature point data from T265.
Will this functions be available in the future?

antithing, 2019-04-25:
+1 for this. Would be very useful for multiple camera tracking configurations.

bfulkers-i, 2019-05-10:
Although we don't yet have exposing this functionality on our roadmap we do like the suggestion. In the meantime, if you have host compute available you may be interested in https://github.com/IntelRealSense/librealsense/pull/3951 which is a python example that uses the T265 intrinsics and extrinsics to compute a stereo image in OpenCV.

RealSenseSupport, 2020-09-09:
Thank you for highlighting this new feature. We have moved our focus to our next generation of products and consequently, we will not be addressing this in T265.

closed 2020-09-16 by RealSenseSupport