realsenseai/librealsense #3630: Why is there nothing in the module that I have already followed the steps?Help!

zh-12345, 2019-03-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
[image]

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@zh-12345  Could you please provide more details about your issue?
1. Librealsense and FW version
2. Linux, windows or android?
3. The sample you run and the issue details

zh-12345, 2019-04-09:
> @zh-12345 Could you please provide more details about your issue?
> 
> 1. Librealsense and FW version
> 2. Linux, windows or android?
> 3. The sample you run and the issue details

i used python 3.6 and v2.17 on windows . i showed the code examples as follows . I cannot call a module in the SDK. The module is empty
[image]

closed 2019-04-11 by zh-12345

Harleens13, 2020-07-07:
Hello @zh-12345 ,
How did you get around this issue? Could you kindly post what you did to solve it?