realsenseai/librealsense #3631: What are the "verts" array values in code opencv_pointcloud_viewer.py

Levi568, 2019-03-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 16}  | 
| Kernel Version (Linux Only)    |  (16)                                         | 
| SDK Version                            |  { }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I am trying to extract the x,y,z values of each 3D streaming pixel and tried running the code in "librealsense/wrappers/python/examples/opencv_pointcloud_viewer.py" , and print out the "verts" in line 297. It turns out printing arrays corresponding to the sensor streaming 3D image. I am wondering what do the array values mean(in inch, centimeter or meter?) and how do I specify the x,y,z values I mentioned? I assume that it is x,y,z from left to right but why only x values keep changing but not y ?  And I also notice that there are some zero values even I face the RealSense toward the wall, which should not contain blocking issue.

[image]
Thank you

ogoshen, 2019-04-02:
The vertices are in [point coordinates](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates).

The array is reshaped, but actually comes in a shape of `(h, w, 3)` , when printing you'll see the (inner) 2nd dimension (the Xs) changing faster.
You're also probably seeing all those zero values due to how numpy summarizes when printing large arrays.

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi Levi568,

Based on what ogoshen commented out, do you still need help?
And another thing you may check is converting the big array to a list and then print. You should see more info of the verts.

Thanks!

RealSenseCustomerSupport, 2019-04-18:
----------------------------------------------

Hi Levi568,

If there is nothing else, this one will be closed soon.

Thanks!

Levi568, 2019-04-18:
No, thank you!

closed 2019-04-25 by RealSenseCustomerSupport