realsenseai/librealsense #3632: Get the  depth data from .raw and What's the use of .csv file?

Core055, 2019-03-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435} | 
| Operating System & Version |   {MacOS  QT  | 
| SDK Version                            |  { legacy / 2.<16>.<5> }                          | 
| Language                            |  {C/C++ }                          | 

is the .raw save the depth data?
I can use the following code to get the  depth data:

`#define _CRT_SECURE_NO_WARNINGS
#include<iostream>
#include<opencv2/opencv.hpp>
using namespace std;
using namespace cv;
int main()
{
    int IMAGE_WIDTH = 424, IMAGE_HEIGHT = 240;
    FILE* fp;
    char* imagedata;
    int framesize = IMAGE_WIDTH*IMAGE_HEIGHT;
    fp = fopen("C:/personal document/SegmentationAlgorithm/grabcut/GrabcutTest/record/frame1/d_Depth.raw", "rb");
    imagedata = (char*)malloc(sizeof(short)* framesize); fread(imagedata, sizeof(short), framesize, fp);
    cv::Mat depthMat(Size(IMAGE_WIDTH, IMAGE_HEIGHT), CV_16U, (void*)imagedata, cv::Mat::AUTO_STEP);
    printf("Depth RAW Central Value: %d mm", depthMat.at<short>(IMAGE_HEIGHT / 2, IMAGE_WIDTH / 2));
    fclose(fp); free(imagedata);
    return 0;
}`

First,I want to ask whether this is right.Do I have to get the .bag file?
Second, when use the `rs-save-to-disk` to get the `.png` file and `.csv` file.   What's the use of `.csv` file??

Thank you!

MartyG-RealSense, 2019-03-31:
A .raw image stores depth data, yes.

The .csv file stores depth data in a text format that is suitable for import into database and spreadsheet programs.

The subject of converting a Kinect .dat file to csv was recently discussed in the link below:

https://github.com/IntelRealSense/librealsense/issues/3491#issuecomment-472738208

Core055, 2019-03-31:
Thank you !
you help me a lot

MartyG-RealSense, 2019-03-31:
You are very welcome.  I'm glad I could be of help!  :)

Core055, 2019-04-01:
@MartyG-RealSense , Could you please tell me what's the software you used to open the .csv file?

MartyG-RealSense, 2019-04-01:
I use a free office suite called Apache OpenOffice (a free equivalent to Microsoft Office) that includes 'Calc' (a spreadsheet program equivalent to Microsoft Excel).  A .csv file can be used with that.

https://www.openoffice.org/

Core055, 2019-04-01:
why my result is different from yours?  where is the metadata?
[image]

MartyG-RealSense, 2019-04-01:
Maybe the difference is in the process used to create the .csv file.  The .csv image I showed above was created with the following steps.

1.  Load a Kinect **.dat** format file into OpenOffice Calc as a .txt file.
2.  Re-save the .txt file as a .csv in OpenOffice Calc using the 'Text CSV' menu drop-down on the Save window.

Here's a link to the discussion that my .csv image came from.

https://github.com/IntelRealSense/librealsense/issues/3491

Core055, 2019-04-01:
okay...my .csv file is saved by `rs-save-to-disk.cpp` , and by the `realsense-viewer`'s snapshot function  have the same result . Maybe I should find another way.

MartyG-RealSense, 2019-04-01:
Hopefully somebody on this forum can provide you with some further suggestions.  Good luck!

ev-mp, 2019-04-22:
@Core055  hi, sorry for the late responce.
The `csv` produced by the `rs-save-to-disk` demo hold the frame's metadata attributes ([documentation](https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk))
It is designed to satisfy the constrains of headless systems with no or very limited terminal bandwidth for visualization. Storing metadata attributes into CSV file provides a convenient way to transfer and inspect the results in offline mode.

ev-mp, 2019-05-01:
@Core055 , do you need assistance with the documentation?

Core055, 2019-05-20:
@ev-mp ,thank you , I have the .raw file to get the depth data , so I dont need the .csv.Thank you again!

closed 2019-05-20 by Core055

kyranf, 2019-05-20:
Ugh.. it hurt to read this thread.. Sometimes help can be misleading/miss the target. 

@ev-mp   there seems to be a way to automatically save the metadata, or at least the Realsense Viewer  and rs-save-to-disk demo uses lets say a 'standard' way of doing it. 

Is there not a simple utility function to point at a  metadata.csv file and have it automatically populate an  rs2_frame_metadata struct for the purpose of offline loading of images captured with the Realsense Viewer?  The images might not have the same image width/height and the intrinsics will be different if the images come from different devices. 

About 1.5 years ago I wrote my own intrinsics serializer and deserializer to store this kind of thing for offline re-running and development, but  that was with my own captured and pre-processed data, not quite the same as what is output by the Realsense Viewer application. 

Any tips?

ev-mp, 2019-05-20:
@kyranf hello, imho the essential question is how the meta/depth data shall be used once it is generated, as this directly derives the requirements for the output format:
- In case the main use-case is post-mortem analysis or  testing then making the output as human and machine-friendly as possible is highly recommended. SDK delivers with `rs-save-to-disk` and `realsense-viewer` snapshot functionality. Both tools are designed to generate  ASCII/RAW format that can be parsed and manipulated directly with Matlab/OpenCv/Excel tools.

- If the data shall be replayed repeatedly and as a sequence, i.e. where the time and effort invested into de-serialization matters, then stream-oriented formats such as SDK's ROSBag imo have a clear advantage even though parsing the data requires non-standard tools. 
ROSbag records preserve not only meta/video data but also the sensor parameters, such as extrinsic/intrinsic into a single indexed database

As to the other part- it is possible to `recreate` the original frame with its attributes given that all the original attributes were somehow preserved. Moreover, this method, that involves `software_device` object is actively employed in several applications. 
This class was designed to serve as an `gateway` for integrating external sensors/data formats with the SDK.  Therefore it does not constrain the inputs to any specific format but rather gives the user full control on how the external data shall be imported/processed.

In case you think to follow this up please consider opening a ticket - replaying to a closed issue is less visible and also counter-productive.

sayyidabeegam, 2020-09-18:
> A .raw image stores depth data, yes.
> 
> The .csv format stores depth data in a text format that is suitable for import into database and spreadsheet programs.
> 
> [image]

How did you saved this depth data as csv file.rs-save-to-disk is saving metadata only in csv format