realsenseai/librealsense #3633: Obtaining IR left and right

spaltiel, 2019-03-31:
| Camera Model                       | D435
| Firmware Version                   | 5.11.01.100
| Operating System & Version | Win 10
| Platform                                 | PC
| SDK Version                            |  2.18
| Language                            |  C#

Hi,

We have the following code (from previous which was working):
var colorFrame = frames.getColor().DisposeWith(frames);
var depthFrame = frames.getDepth().DisposeWith(frames);
var IR0Frame = frames.getInfrared(1).DisposeWith(frames);
var IR1Frame = frames.getInfrared(2).DisposeWith(frames);

However now 'getDepth', 'getColor', 'getInfrared(i)' are not working anymore. We guess it was deprecated.
We also noticed that:
 getColor replaced with ColorFrame
 getDepth replaced with DepthFrame
 getInfrared replaced with InfraredFrame

We could not find how to get both Left and Right.

How it is recommended to get these IR frames?

Regards,
Amir

MartyG-RealSense, 2019-03-31:
To display both the left and right IR frames, you should use an index number to differentiate between them.  The link below has an example script for the 400 Series cameras under the 'Librealsense2' heading for streaming the left and right IR channels.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers

spaltiel, 2019-03-31:
Dear Marty,

Yes we did and do the following:
var cfg = new Config();
cfg.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 30);
cfg.EnableStream(Stream.Color, 1920, 1080, Format.Rgb8, 30);
cfg.EnableStream(Stream.Infrared, 1, 1280, 720, Format.Y8, 30);
cfg.EnableStream(Stream.Infrared, 2, 1280, 720, Format.Y8, 30);
pipelineProfile = pipeline.Start(cfg);

...

later we need to actually get the two IR frames.
In the past we used:
frames.getInfrared(1).DisposeWith(frames); 
frames.getInfrared(2).DisposeWith(frames);
which return both left and right images.

However, in the new driver frameset does not contain definition for 'getInfrared' anymore and the new definition 'InfraredFrame' does not expose indexes (at least we did not find how) - we aactually asked where this indexes for left and right are now available in the new driver.

In the suggested reference you mentioned we did not find a solution to our question. 
Sorry in advanced if we miss it.

Regards,
Amir I.

MartyG-RealSense, 2019-03-31:
Sorry, I did not see on first reading that you were using C#.  The number of available references for C# versions of scripting is limited.  For now until one of the Intel guys on this forum can follow up with you, the best I can suggest is to look at the link below of a C# code snippet and see if you can adapt it to show the two IR streams.  I do apologize.

https://github.com/IntelRealSense/librealsense/issues/1747#issuecomment-391636433

spaltiel, 2019-03-31:
thanks!
Indeed in C++ we mange to get it.
However, we also need it for C#. We do want to use Intel wrapper and do not try to wrap it our self.

ogoshen, 2019-04-01:
```cs
var left = frames[Stream.Infrared, 1];
var right = frames[Stream.Infrared, 2];
```
Indeed missing in docs, should be added to the [cookbook.md](https://github.com/ogoshen/librealsense/blob/4b7b6c9cbcc68e96f840e13255924ca2f8e426a7/wrappers/csharp/Documentation/cookbook.md) in #3602 

Will consider adding a `GetInfrared(int index = 0)` method too, since the property access doesn't support parameters. 

Most of these methods are eventually just this:
```cs
FirstOrDefault<T>(Predicate<T> predicate) where T : Frame
```
with the predicate filtering for stream type, format, index etc.

spaltiel, 2019-04-02:
Omer thanks!

`var left = frames[Stream.Infrared, 1];`
`var right = frames[Stream.Infrared, 2];`
indeed it solved the issue.

As you suggested if you will make "GetInfrared(int index = 0)"  it will be optimal (at least it is most intuitive for us).

Cheers,
Amir I.

closed 2019-04-02 by spaltiel