realsenseai/librealsense #3635: How to write .raw file and .csv file by OpenCV or by "fstream.h"

Core055, 2019-03-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (5.10) | 
| Operating System & Version |   {MacOS  | 
| SDK Version                            |  { legacy / 2.<16>.<5> }                          | 
| Language                            |  {C/C++ }                          | 
 
I can write color and depth image successfully,using the following code:
`rs2::pipeline p;`
`p.start();`
`for (auto i = 0; i < 30; ++i) `
`	p.wait_for_frames();`
`rs2::frameset frames = p.wait_for_frames();`
``
`auto vf = frames.as<rs2::video_frame>();`
``
``
`rs2::frame colorFrames = frames.get_color_frame();`
`const int cw = colorFrames.as<rs2::video_frame>().get_width();`
`const int ch = colorFrames.as<rs2::video_frame>().get_height();`
`Mat colorImage = Mat(Size(cw, ch), CV_8UC3, (void*)colorFrames.get_data());`
`cvtColor(colorImage,colorImage,CV_BGR2RGB);`
`        `
`rs2::frame depthFrames = frames.get_depth_frame();`
`static rs2::colorizer color_map;`
`const int dw = depthFrames.as<rs2::video_frame>().get_width();`
`const int dh = depthFrames.as<rs2::video_frame>().get_height();`

`static rs2::frame color_depth_frames;`
`color_depth_frames = color_map.colorize(depthFrames);`
`Mat depthImage = Mat(Size(dw, dh), CV_8UC3, (void*)color_depth_frames.get_data());`

`cv::imwrite(desktop_path.toStdString()+"/"+std::to_string(c)+"-depth.png",colorImage);`
`cv::imwrite(desktop_path.toStdString()+"/"+std::to_string(c)+"-color.png",depthImage);`

when I try to write .raw file and .csv file using the method mentioned in the SDK `save_frame_raw_data` ,`metadata_to_csv`  ,   the problem comes:
 
[image]

Thanks !

dorodnic, 2019-04-01:
Hi @Core055 
Please take a look at [rs-convert](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert) and [save-to-disk](https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk) implementation. `save_frame_raw_data` is not part of SDK API.

Core055, 2019-04-01:
I will try !  
Thank you !

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@Core055 Did you get it through? Any other questions about this ticket? Looking forward to your update. Thanks!

Core055, 2019-05-20:
OK, I've gotten it through!thank you for asking!

closed 2019-05-20 by Core055