realsenseai/librealsense #3636: cv2.imshow is showing blank white screen

gauravbansal98, 2019-03-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2019-04-01:
Hi @gauravbansal98 
This is a long-shot, but what is your firmware version? I think there were some such issues with 5.8.x firmware...

gauravbansal98, 2019-04-01:
firmware version is 5.10.13

RealSenseCustomerSupport, 2019-04-24:
----------------------------------------------

Hi @gauravbansal98,

Is your camera enumerated as a USB2 device? Please try to remove below line to see how it works.
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

Hi @gauravbansal98,

Any update? If this is clarified on the Raspberry Pi 3, I will close this.

closed 2019-11-01 by RealSenseCustomerSupport