realsenseai/librealsense #3637: T265 Camera jitter when stationary

yingbo-chen, 2019-04-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { t265} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2  | 
| Language                            |  {python}                          | 
| Segment			|  VR/AR/                         | 

### Issue Description
T265 Camera jittering quite a lot when fixed to a surface with no movement and vibration, is there any way to fix the jitter mostly in Z axis.

SlavikLiman, 2019-04-01:
Hi @yingbo-chen ,
Please make sure that you're using the latest libRealSense version (2.19.2).
1. Is the amount of jitter constant?
2. How bad is it? (please share values)

yingbo-chen, 2019-04-02:
I have updated to 2.19.2, no real improvement to jitter issue

Looking at the realsense viewer the value of x y z fluctuate by around 0.001 to 0.002 on all axis a couple times every second.

I have also discovered that in the exact same environment and lighting, the sensor starts with medium confidence and after a while it switches to high confidence. what's affecting the confidence? 

Also sometimes relocation starts looping every seconds with no movement.

ShettyHarapanahalli, 2019-04-02:
even I have observed jitter when using it.

bfulkers-i, 2019-04-03:
Hi @yingbo-chen, thanks for checking `2.19.2`. We are continually updating the firmware for T265 to try to improve quality and robustness for common applications. The fluctuations you see when stationary seem to be about 1mm. Can you describe briefly the application where this is significant so we can add it to our list to keep in mind? 

The confidence should not switch to high while stationary, high confidence implies that the device has undergone enough motion to help T265 understand its local position accurately. If you see the device switch to high confidence after starting without moving it, please open a separate issue to track that.

Currently, you should not see relocalization events in `2.19.2` (due to #3477) while we work on a release to improve their usefulness.

yingbo-chen, 2019-04-03:
currently im using it for VR and AR applications.
When starting the camera, should I move the camera around first to initialize it's surrounding for it to be more accurate?

bfulkers-i, 2019-04-04:
Thanks @yingbo-chen. For VR/AR what we have seen is that the two most important things to reduce perceived jitter are prediction and correctly accounting for extrinsics between the tracking origin on the device and the expected tracking origin in your rendering system. Assuming you are already doing the latter, here is an example showing how to do prediction that may be helpful to you:

https://github.com/IntelRealSense/librealsense/tree/master/examples/pose-predict

yingbo-chen, 2019-04-05:
Thank you @bfulkers- . I'm fairly inexperienced with VR/AR development, how does prediction work?
the example given in 
```
rs2_pose predicted_pose = predict_pose(pose_data, dt_us);
```
what is dt_us? Im currently using python wrapper
```        
pose = frames.get_pose_frame()
        if pose:
            data = pose.get_pose_data()
```
do I just call predict_pose(pose_data,dt_us) to get a new set of predicted pose?

Im also too sure of what do you mean by correctly accounting for extrinsic between virtual camera and the sensor, could you explain a bit more on how I could improve the jittering?

RealSenseCustomerSupport, 2019-04-05:
----------------------------------------------

@yingbo-chen - I believe what @bfulkers-i meant by extrinsics with regards to AR/VR is there is a center of tracking for the T265 device, when in AR/VR scenario you should take into consideration the location of the center of tracking of T265 with the location of the "center" in VR space.  The T265 would be mounted outside on a VR/AR headset or device, translating using extrinsics or mechanical location of center of tracking with regards to "center" of VR/AR device location is necessary.

There is a brief explanation in the datasheet:  https://dev.intelrealsense.com/docs/datasheets
The datasheet has a mechanical diagram with extrinsics showing center of tracking location as well as outlines tracking coordinate system.

closed 2019-05-23 by RealSenseCustomerSupport