realsenseai/librealsense #3638: Import pyrealsense2 : OK in terminal / No module named 'pyrealsense2 in an IDE

SamConti, 2019-04-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | v2.19.2 | 
| Operating System & Version |   Ubuntu Mate 16.04.2 | 
| Kernel Version (Linux Only)    |  4.4.38-v7+                                         | 
| Platform                                 | Raspberry Pi3  |
| Language                            |  python                        | 

Hello,

I install the realsense SDK package to use a D435 camera. All seems to be correctly installed:

- Intel RealSense SDK2.0
- Python Bindings
- Python Path

It's from point 10, 11, 12 and 13 in this procedure https://github.com/PINTO0309/MobileNet-SSD-RealSense/blob/master/README.md

I'm able to import pyrealsense2 librarie in a termlinal when I launch Python3 (no error) but when I want to import it in a python script in Eric IDE (or another IDE) I have : 'Import Error : No module named 'pyrealsense2'.

Any idea how to work this libbarry with an IDE as Eric ?

Thank you,

SamConti, 2019-04-01:
Adding librealsense.so and pyrealsense2.cpython.35m-arm-linux-gnueabihf.so resolves the problem.

closed 2019-04-01 by SamConti

chaitudarz, 2019-04-23:
Hello,

I have followed the installation procedure from https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md. 

ubuntu 16.04.15 LTS 
python3
sdk2.0
platform Laptop

I am facing a similar error when I execute the python files in IDE ("Import Error: No Module named Pyrealsense2). I have no other issues in the normal terminal.

And I even cannot find these files in my python wrapper
[image]
 "**librealsense.so and pyrealsense2.cpython.35m-arm-linux-gnueabihf.so**"" . 

Any ideas or suggestion to make it work?

Thank you

kat-mcg-lin, 2019-05-27:
@SamConti @chaitudarz 
Have you both figured it out? I have been trying as well and it only works in the terminal for me when I open python up in the exact location of the wrapper, but won't work in pycharm

If you have figured it out, could you please share? thanks!