realsenseai/librealsense #3640: Extract data from PNG Depth images

Amahmoud1994, 2019-04-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 / D400 } | 
| Firmware Version                   | (v2.13.0) | 
| Operating System & Version |   {Linux (Ubuntu 16.04)}  | 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {others }                          | 

### Issue Description

I have PNG depth images without any extra information about them {e.g no .RAW files} .. Is it possible to extract depth data from them? and if not, is there any data that can help me in the same direction?

Amahmoud1994, 2019-04-01:
[image]

This is a sample of the images I have

dorodnic, 2019-04-01:
This is somewhat repeating question. The short answer is no, you cannot. 
Please see [record-playback](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback) example for a working method of collecting data.

Amahmoud1994, 2019-04-01:
@dorodnic  I checked the example you mentioned, thank you for that ... 
I had the realsense fixed in one position for all the recordings I did so what if I have a metadata file for one Depth image from these recordings? will this be of help or the answer will still be no?

dorodnic, 2019-04-01:
The main issue is that the colors are not linear in depth, and depend on the histogram of the original image. So red may be 1 meter in one image and 10 meters in another

Amahmoud1994, 2019-04-01:
@dorodnic oh I see, even if it is the same setup\environment in all images?

dorodnic, 2019-04-01:
Pretty much. The closest object will always be blue, the furthest away object will always be red and the median (not the average) depth will be yellow. So unless it is the same scene exactly, there is no way to know. Sorry

Amahmoud1994, 2019-04-01:
@dorodnic ok thanks a lot :+1:

closed 2019-04-01 by Amahmoud1994