realsenseai/librealsense #3642: D415 structure light

czjczjczj1, 2019-04-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2019-04-01:
This might help:
https://github.com/IntelRealSense/librealsense/blob/master/doc/depth-from-stereo.md
The camera "sees" both IR and visible light and performs dense stereo matching based on features it can see. In visible wavelength, such features would be visible features, like corners, texture, edges, etc...
In infrared, it also can pick up on artificial features generated by the projector (in addition to natural features)

closed 2019-04-01 by czjczjczj1

marsarCV, 2021-01-27:
> This might help:
> https://github.com/IntelRealSense/librealsense/blob/master/doc/depth-from-stereo.md
> The camera "sees" both IR and visible light and performs dense stereo matching based on features [...]

How is Active IR Stereo different than Coded light(as in Intel SR300)?