realsenseai/librealsense #3644: D435i not recognized by any SDK examples or RealSense Viewer

Brekel, 2019-04-01:
-------------------------------------------
| Required Info
-------------------------------------------
| Camera Model               | D435i      | 
| Firmware Version           | 5.11.1.100 | 
| Operating System & Version | Win 10     | 
| SDK Version                | 2.19.2     | 
-------------------------------------------

### Issue Description

My D435i doesn't seem to get recognized by RealSense Viewer (or any other SDK example for that matter)
Windows does seem to recognize it as it shows up in Device Manager when plugged in.
I've tried two different D435i sensors, on the same machine with the same (supplied) cable, several different D435 and D415 sensors do work correctly.
I've tried two different machines, a laptop (Dell XPS15) and desktop machine, on the desktop machine both on the on board USB3 and a PCI Express USB3 card.
I've tried using the SDK release package as well as rebuilding from source.
I've tried CustomRW from the calibration API to 'gold reset' the camera to factory default as outlined below:
https://forums.intel.com/s/question/0D50P00004FPDzwSAH/realsense-d435i-not-properly-recognized-by-realsenseviewer?language=en_US

The rs-enumerate-devices example simply displays:
No device detected. Is it plugged in?

Here's a quick screen recording of me plugging a D435 and D435i whilst Device Manager and RealSense Viewer is open:
https://t.co/S63KVp6BVP

Any idea?

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Hi Brekel,
The SDK 2.19.2 should work with D435i.
Thank you so much running the CustomRW -g in attempt to restore the calibration table. Could you please also help and export the tables using the following commands so we can further check the calibration data?
`Intel.Realsense.CustomRW.exe -r -raw 19 -f table19.bin`
`Intel.Realsense.CustomRW.exe -r -raw 1F -f table1F.bin`
`Intel.Realsense.CustomRW.exe -r -raw 20 -f table20.bin`
If you happen to have another D435i, please give it a try.

dorodnic, 2019-04-02:
In addition, you mentioned that the camera is recognized in device manager. The D435i supposed to appear as 3 distinct devices - two cameras (depth and RGB) and motion sensor, like this:
[image]
Could you please check if you see anything under Sensors when you connect the camera

Brekel, 2019-04-02:
Thank you for the suggestions!

Yes I have 2 D435i devices and both show the same behaviour on 2 machines.
Also both do show the "HID Sensor Collection V2" in Device Manager.
[image]

I've exported the tables from both:
[838212074358.zip](https://github.com/IntelRealSense/librealsense/files/3033683/838212074358.zip)
[841512071761.zip](https://github.com/IntelRealSense/librealsense/files/3033684/841512071761.zip)

One peculiar new observation.
My desktop machine has a USB2.0 port, if I connect it to that just to test it does show up something in RealSense Viewer:
[image]

If there's anything else I can try please let me know how I can help.

dorodnic, 2019-04-02:
If you open Viewer settings page (button on the top right), under General tab, you can enable and set logging preferences:
[image]
Please enable logging to file and set the verbosity to DEBUG and attach the resulting log after connecting the camera

Brekel, 2019-04-02:
Sure, here's a session with connecting to USB3 (verified the 3 items in device manager to appear), then disconnect, then closing the viewer.
[librealsense.log](https://github.com/IntelRealSense/librealsense/files/3034008/librealsense.log)

dorodnic, 2019-04-02:
Thanks @Brekel 
Sorry for the trouble with the device. 
This - `"This program is blocked by group policy. For more information, contact your system administrator."` looks very suspicious, and seem to prevent enumeration of motion sensors. 
Is it possible to try running the App as administrator?

Brekel, 2019-04-02:
No worries on the trouble, I'm here to help debug :)
Here's another fresh log when running the viewer with admin rights:
[librealsense.log](https://github.com/IntelRealSense/librealsense/files/3034060/librealsense.log)

dorodnic, 2019-04-02:
Reading-up on the error, it doesn't look good. 
I have never encountered this issue before, so I'm not 100% sure what will help.
However, the following links looked most promising to me:
* [this-program-is-blocked-by-group-policy-error-windows-10/](https://www.isunshare.com/blog/this-program-is-blocked-by-group-policy-error-windows-10/)
* [214461-local-group-policy-reset-default.html](https://www.sevenforums.com/tutorials/214461-local-group-policy-reset-default.html)
* [programs-blocked-by-group-policy-but-why](https://social.technet.microsoft.com/Forums/windows/en-US/5831722b-3f5e-4834-985a-6f9c014c9bd4/programs-blocked-by-group-policy-but-why)

Brekel, 2019-04-02:
Thanks for the tip but I strongly doubt that's it as my D435 and D415 models work on that machine without a problem. Even when viewer is not running in admin mode.

Actually I just tried 2 other machines, one shows the exact same problem.
However both the D435i's work fine on an Intel NUC.
Maybe it could have something to do with the USB chipset?

dorodnic, 2019-04-02:
The thing is - SDK recognizes depth and RGB parts of the D435i just fine, but it doesn't show up in the Viewer because we cannot enumerate motion sensors, and this is the only part failing on `This program is blocked by group policy` error. D435 / D415 works because it doesn't have a motion sensor. 
I'm fairly confident this is purely OS related.

Brekel, 2019-04-02:
Ahaa, thanks to that last remark I found the culprit and got it working! :)

Turns out in an effort to shut down Microsoft sending way too much telemtry data home a program called "Spybot Anti-Beacon" was installed to perform some registry options for that.

I narrowed it down to two registry items it set in order to enable/disable what it calls "sensors":
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\DeliveryOptimization\Config\DODownloadMode
Computer\HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\DeliveryOptimization\SystemSettingsDownloadMode

If I disable those in that app things work as expected!

closed 2019-04-02 by Brekel

ev-mp, 2019-04-02:
@Brekel , this indeed seem to be OS-related as noted by @dorodnic .
Please follow the instructions in the link to enable MS location services:
https://stackoverflow.com/questions/45329026/sensor-diagnostic-tool-wont-run
then retry D435i.

To verify that the Windows OS is properly configured for Accel/Gyro sensor you can also use a genuine MS utility for HID sensors called `sensordiagnostictool.exe` distributed as [part of WindowsSDK 10](https://social.msdn.microsoft.com/Forums/en-US/03a719ec-5c67-4ede-8cc9-32cf9d1979ad/how-to-get-sensor-diagnostic-tool-?forum=windowssensorandlocationplatform)
If that utility load and run properly with D435i, then the device should also be correctly enumerated in Librealsense, and vice versa.

Brekel, 2019-04-02:
@ev-mp thanks for the suggestion, as I mentioned in the last post it got resolved.

dorodnic, 2019-04-02:
Glad it worked!
Feel free to open new issues if you have any follow-up questions / issues with the camera

Brekel, 2019-04-02:
Thanks for all the help!
I've been working with RealSense since the 200 series, this was my first issue, great sensors and SDK! :)

I am thinking about maybe doing a pre-check in my apps for these types of settings by looking at the registry values at startup to see if things are blocked. May be an idea for the RealSense viewer as well?

dorodnic, 2019-04-02:
I'm very cautious doing anything in the Viewer that might go beyond minimal required permissions, like looking at the registry. However, we might need to expose some type of software interface to register to warnings during enumeration process (we have such notifications API once you have the camera object). 
Maybe provide a script, similar to [the hardware-timestamp script](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense_metadata_win10.ps1)...
Also, for cameras we usually have a backup - on Linux that's `FORCE_LIBUVC` and for Windows its `FORCE_WINSUB_UVC`. For now, we don't have such backup for Motion Sensor and rely entirely on Microsoft Sensor framework. There is a background task in progress to implement alternative method (this would, for instance, also solve the issue for Windows 7)

Brekel, 2019-04-02:
I was just thinking out loud and am sure you can come up with a much more clever and cross platform solution than I would :)

Ajasra, 2019-09-05:
this solution helped me. Delete all existing group policies.
https://appuals.com/this-program-is-blocked-by-group-policy/

ll7, 2020-04-08:
The problem for me was, that the location privacy settings are managed by my network administrator. It is not intuitive that the camera needs location privacy settings enabled.
Anyway, the discussion helped me. Thank you.

makhija, 2020-10-23:
I notice this access problem with D435i sensor however not in real sense viewer application. I developed a .Net application which works well D435 & D415 sensor, but runs into an error  when using D435i sensor. Interestingly the error occurs well after I have successfully accessed the depth and colour images and proceed to call a C++ method from C#. 

Error message: "System.AccessViolationException: 'Attempted to read or write protected memory. This is often an indication that other memory is corrupt.'

In the code below, i get the exception at:  _measurer.Measure(dataptr, regions, _calibrationData);

// K new 
                        idepthFrame.CopyTo(depthdata);
                        unsafe
                        {
                            // get access to data and then see the followin convertion to dataptr
                            _calibrationData.angleCameraTorsoRadians = 0.43;
                            fixed (ushort* dataptr = depthdata)
                            {
                                _measurer.DepthMethod = Method;
                                _measurer.Measure(dataptr, regions, _calibrationData);
                            }                          
                        }