realsenseai/librealsense #3645: Depth Quality Tool (d400)- Corrupted settings - Emitter enabled value

StevenSpringlab, 2019-04-01:
| | |
|---|---|
|**librealsense**|2.19.0 RELEASE|
|**OS**|Windows|

Situation:
Using the 2.16.1 version of the Depth Quality Tool, the tool crashed while trying to change the emitter enabled value.
Tested using 2.16.1 and 2.19.0 version, on a W10 computer with now corrupt emitter value and on a W10 computer without errors.
The camera's used are d435's running on firmware 5.10.6 (on computer with corrupted settings) and 5.11.1.100 (on other computer), however the firmware version seems unrelated as I've used the 5.10.6 with the tool repeatedly in the past (occasionally getting stuck/crashing, but nothing rebooting the tool didn't fix at the time).

Issue:
After the crash, the settings were saved in a corrupt state. Now showing the emitter enabled value as a slider, rather than a checkbox, while giving the following error message:
'Invalid Value in rs2_get_option(options:0A44318, option:Emitter Enabled); get_xu(id=2) failed! Last Error: No error'

Steps taken:
- Turn the computer off and on again
- The slider shows the value set to 0. I tried setting it to 1, both via sliding and via the text input.
This same error occurs using both 2.16.1 as well as in 2.19.0.
- Trying to reset the values to default, via the Presets or via Options - Settings - General, then Restore Defaults was of no avail, as even on the properly functioning computer, this doesn't reset the Emitter Enabled value.
- Changing the preset to Default (or via the settings using Restore Defaults) doesn't change the Emitter Enabled or the Histogram Equalization Enabled states. And presumably some other settings as well.

Idea for generic workaround:
Not sure it's intentional that only part of the settings are stored, but some indication which values are stored might be beneficial, as well as an option to fully reset the tool to it's initial state.

Minor side note:
While the Realsense camera isn't connected, the message on screen is 'Realsense device detected.', rather than the 2.16.1 version where it flashed 'Please connect Intel Realsense device!'

RealSenseCustomerSupport, 2019-04-10:
----------------------------------------------

Hi StevenSpringlab,

I tried with a D435 (FW5.10.6) and have no issue when using Depth Quality Tool (both 2.16.1 and 2.19.0).
My system is Windows 10 version 1709 Build 16299. Please see the attached screenshots.

Can you please share some screenshots and also wonder what are the differences between your Windows 10 machines (showing the corrupted issue vs. no issue)?

Thanks!

Attachment(s):
dqt_2.19.0_emitter_off.png - https://realsensesupport.intel.com/attachments/token/BQxEug91ppHbTPh78QV4lr5h0/?name=dqt_2.19.0_emitter_off.png
dqt_2.19.0_emitter_on.png - https://realsensesupport.intel.com/attachments/token/Ua9cNnwDqCeyDKWV8WWMcWBUl/?name=dqt_2.19.0_emitter_on.png
dqt_2.16.1_emitter_on.png - https://realsensesupport.intel.com/attachments/token/yGIa1HkRu5SVitqtoAA8lhjwf/?name=dqt_2.16.1_emitter_on.png

StevenSpringlab, 2019-04-10:
Thank you for your reply.
After a reply on the forum thread, I updated the firmware just now and this fixed the problem.

To expand a bit on where the bug came from, the problem wasn't with the firmware or Depth Quality Tool versions themselves, as those worked fine in the past. The emitter value only got corrupted when the program hung up and was terminated manually. In the issue I called this a crash, but I realise now this disregards some relevant differences results.
I hadn't expected the value to be stored in the camera itself, though updating the firmware should've been a step to try sooner.

While no longer relevant, I should also note that I am unable to view the provided attachments, since this requires me to sign in with an Intel Business account.

closed 2019-04-10 by StevenSpringlab