realsenseai/librealsense #3648: How to built librealsense on Windows 10?

AngeloEspinoza, 2019-04-01:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 | 
| Operating System & Version |   {Win (8.1/10) }  | 
| Language                            |  {Python}                          | 
| Segment			|  {Robot }                          | 

I'm trying to install librealsense on Windows 10 but the information is not that precise, I tried with the Cmake tutorial but I didn't understand it, now I'm trying to understand how to built the librealsense on windows here: https://ci.appveyor.com/project/ddiakopoulos/librealsense, however I'm not able to understand how to built it since I just see a bunch of code but I don't understand how to built it, I'd appreciate some help.

dorodnic, 2019-04-02:
There are Visual Studio 2015 and 2013 solution files in the old branch:
https://github.com/IntelRealSense/librealsense/blob/legacy/librealsense.vc14/realsense.sln
You should be able to just open it and build.

AngeloEspinoza, 2019-04-02:
What do you mean? I don't get it, should I copy that in a file .sdl and build it or what should I do? sorry for the questions I'm new at this

dorodnic, 2019-04-02:
Download and unzip [legacy.zip](https://github.com/IntelRealSense/librealsense/archive/legacy.zip)
Navigate to `librealsense.vc14` folder and open `realsense.sln` using Visual Studio 2015. If you don't have Visual Studio, I believe Visual Studio Express can be used. Then, just build the project...

AngeloEspinoza, 2019-04-02:
It worked, thank you very much, now I'm trying to install the Python wrapper and I'm following the steps as in [](https://pyrealsense.readthedocs.io/en/master/readme.html), I'm setting the environment variables in the system with the directory pointing to the folder 'include' where is the file 'rs.h' (all in the unzipped folder that you passed me), and  I'm trying to install pyrealsense through pip but that's not working I have `LINK : fatal error LNK1181: 
can not open the input file 'realsense.lib'`, I tried some examples in visual studio and they're actually running but I cannot install the Python wrapper I'd appreciate you can help me with this

dorodnic, 2019-04-02:
Glad it helped. I'm not experienced with pyrealsense (it is a community project) but I'll try to look at it

AngeloEspinoza, 2019-04-02:
Thank you, I'll be waiting for it

dorodnic, 2019-04-03:
Ok, managed to build it. 

Couple of tricky points:
a. I think you might need to specify build configuration:
[image]
This might be dependent on install python version (Win32 or x64)
b. Don't forget to update `PYRS_LIBS` accordingly (either to end with `\bin\x64` or `\bin\Win32`)
c. Any time you change one of the environment variables you have to close and re-open the command line prompt (for the change to take effect)
d. I didn't understand "You might also need to have stdint.h available in your path". What worked for me is just downloading `stdint.h` from the internet and dropping it to librealsense include folder (next to `rs.h`)

Hope this helps

AngeloEspinoza, 2019-04-03:
Ok, I noticed you took that pic from visual studio but I don't really know why, what is the build configuration for? 
the `PYRS_LIBS` are directioning to the `bin\Win32` of the folder that you passed me? this is: 
[image]?
In c. I guess you're saying that because I'm installig the wapper through pip, right? 

Thank you for the response, and for keep helping me

AngeloEspinoza, 2019-04-03:
Forget about the last answer, I redirectioned the `PYRS_LIBS` to the correct path (_the one that I put) and it finally worked!
[image]
However I tried to import pyrealsense as always `import pyrealsense` in Python to start with using it but I'm getting these errors
` class CParser(PLYParser):
Traceback (most recent call last):
  File "C:\Users\ElectronicsLab02\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pycparser\__init__.py", line 42, in preprocess_file
    text = check_output(path_list, universal_newlines=True)
  File "C:\Users\ElectronicsLab02\AppData\Local\Programs\Python\Python36-32\lib\subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "C:\Users\ElectronicsLab02\AppData\Local\Programs\Python\Python36-32\lib\subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\ElectronicsLab02\AppData\Local\Programs\Python\Python36-32\lib\subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "C:\Users\ElectronicsLab02\AppData\Local\Programs\Python\Python36-32\lib\subprocess.py", line 1017, in _execute_child
    startupinfo)`

I emphasize that I don't have `stdint.h` since I did not find where to download it, can you provide me a link where you download it and the place where you place it, please?

AngeloEspinoza, 2019-04-03:
`RuntimeError: Unable to invoke 'cpp'.  Make sure its path was passed correctly
Original error: [WinError 2` 
I've got this error as well, I'm tying to solve it

dorodnic, 2019-04-07:
I tried importing the module and installing MinGW as suggested in 77, but with no luck. I'm getting the same error as you. Perhaps you can raise it again on pyrealsense forums?

AngeloEspinoza, 2019-04-08:
Ok, I'll do that, hope to find a solution

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@AngeloEspinoza Did you get through the issue? Any other support needed from here? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@AngeloEspinoza Any other questions about this ticket? Please update. Thanks!

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@AngeloEspinoza If no other questions for this ticket, will close it accordingly. Please feel free to create new ticket if you still have other questions. Thanks!

closed 2019-05-16 by RealSenseCustomerSupport