realsenseai/librealsense #3652: T265 integration/demo with robots

ShettyHarapanahalli, 2019-04-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 and D435 | 
| Firmware Version                    
| Operating System & Version    Ubuntu 16
| Kernel Version (Linux Only)   
| Platform                                 | PC/ NVIDIA Jetson   |
| SDK Version                            |   2 }                          | 
| Language                            |  {C/C/python/}                          | 
| Segment			|  Robot                          | 

### Issue Description
Hi,

Do we have an example project/configuration of T265 with move base ROS package? 

Any example configuration for robots like :  https://github.com/NVIDIA-AI-IOT/jetbot or https://github.com/autorope/donkeycar working with this, this will be a very good point to start over. 

BR,
SSH

bfulkers-i, 2019-04-10:
Thanks @ShettyHarapanahalli this is a great idea, we plan to get to this as part of expanding our wheel odometry support in ROS, but it may take a little while to arrive.

ShettyHarapanahalli, 2019-06-21:
I can see the Occupancy mapping in the below project, is there any further work planned on this - 
https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping

RealSenseSupport, 2020-09-09:
Thank you for highlighting the use of T265 with move base ROS package. We have moved our focus to our next generation of products and consequently, we will not be addressing this feature in the T265.

closed 2020-09-16 by RealSenseSupport