realsenseai/librealsense #3653: macOS installation issues

daniel-dennis, 2019-04-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   MacOS (Mojave) | 
| SDK Version                            | 2.0        | 

### Issue Description
I can't install vulkan-sdk, I've tried using `brew cask install vulkan-sdk`, but it doesn't work, it says `Error: Cask 'vulkan-sdk' is unavailable: No Cask with this name exists.`. Is there something I'm doing wrong here? I have homebrew installed.

gravitychen, 2019-04-05:
I have this same problem

jzw0025, 2019-04-05:
I tried this command and it works: brew cask install apenngrace/vulkan/vulkan-sdk

sbartstv, 2019-04-13:
Can't you just make a simple Mac Install file. This is a real pain too many errors.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@daniel-dennis Any other questions about the ticket? Looking forward to your update. Thanks!

sbartstv, 2019-04-26:
I got it all working on Mac thanks. The real problem is lack of free helper tutorials, videos and examples. back a few years ago, you gave out simple starter free software so we could scan on the old units, I have a D415 and I want to do point cloud recordings abc rendered and also just simple good 3D body and head scan, there are a few options to purchase software but this is basic stuff and should be included like you used to as there are not enough developers doing projects. Can you please also point me on good starter code or videos for D415 for computer vision and 3D Scanning, point clouds and volumetric video capture. Thanks.

daniel-dennis, 2019-04-26:
> @daniel-dennis Any other questions about the ticket? Looking forward to your update. Thanks!

That's all fine, thank you.

closed 2019-05-05 by RealSenseCustomerSupport