realsenseai/librealsense #3657: T265 not detected over USB hub

ShettyHarapanahalli, 2019-04-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 / t265 } | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   {Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { 2.19.2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
Hi, 

I am trying to use 2 D435 and T265 for a robot when I connect these over the USB hub I connect it to Jetson Tx2 or on laptop I do not see the T265 detected.     

LSUSB gives me the below result

$ lsusb
Bus 002 Device 009: ID 8086:0b07 Intel Corp. 
Bus 002 Device 008: ID 8086:0b07 Intel Corp. 
Bus 002 Device 007: ID 05e3:0612 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 27c6:5381  
Bus 001 Device 002: ID 0c45:6718 Microdia 
Bus 001 Device 008: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 007: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 004: ID 8087:0aaa Intel Corp. 
Bus 001 Device 012: ID 03e7:2150  
Bus 001 Device 011: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

When I see T265 in sources realsense-viewer I do not see the tracking camera module. 

I see when its directly connected to laptop. 

Do we have any details on the USB hub interface requirements, inducing BW, power etc. when we need to connect multiple devices.

SlavikLiman, 2019-04-03:
Hi @ShettyHarapanahalli ,
Please try the following sequence:
1. Connect the USB hub to the host
2. Connect the T265 to the USB hub
Let me know if it works for you.

ShettyHarapanahalli, 2019-04-03:
Hi, 

The USB hot plug (above sequence) works, but I can not do this on the jetson/robot every time do you know any solution to fix this. 

Best Regards,
Suman

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

Hi

The hub ultimately has to be connected to the system, powered and recognized before the T265 device is connected to the hub itself.  That way the hub and T265 isn't being powered on at the same time and being recognized by the host platform at the same time.

closed 2019-04-26 by RealSenseCustomerSupport

raymondbedwards55, 2019-12-01:
I can't believe that the solution being offered for this issue is to unplug and re-plug the T265 device.  This is a totally ridiculous solution and certainly not one that you could expect any end user to accept.  What plans are there for a real solution to this problem?

MarcoAndrade26, 2020-08-19:
@raymondbedwards55 @radfordi @RealSenseCustomerSupport @ShettyHarapanahalli Could you please help me? I am facing same issue as described here and in this issue I just posted #7125

haljarrett, 2020-08-26:
I see that a recent librealsense release added the hardware reset function to the t265, and that this may be a workaround for the above mentioned issue. Are there any plans to address the underlying issue, or should we plan to work around them in software?

I am working with a t265 in USB 2 mode over an industrial hub in a robotics application. The plug-replug fix seems to work, but is not something we can use in a real environment.

tkazik, 2021-11-22:
This still seems to be an issue :/

Colin-J-Smith, 2023-04-12:
This is still an issue, even with a direct USB connection (no external hub).