realsenseai/librealsense #3658: Python code to save depth images

kongsgard, 2019-04-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1.100 | 
| Operating System & Version | Ubuntu 18.10  | 
| Kernel Version (Linux Only)    | 4.18 | 
| Platform                                 | PC |
| SDK Version                            |  2.19.2 | 
| Language                            | Python | 

### Issue Description
Hi! I have some issues with acquiring depth images from the camera. I've used the Python code included below (a slight modification of the provided example [align-depth2color.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py) which also seems to give the same issue), but I only get images as follows:

[image]
[image]

As you can see, the normal RGB image is fine, while the depth image is very dark. However, in realsense-viewer the depth output looks good.
Any ideas on what could be the cause?

### Code
```
import cv2
import imageio
import numpy as np
import pyrealsense2 as rs
import os
import shutil

def make_clean_folder(path_folder):
    if not os.path.exists(path_folder):
        os.makedirs(path_folder)
    else:
        user_input = input("%s not empty. Overwrite? (y/n) : " % path_folder)
        if user_input.lower() == "y":
            shutil.rmtree(path_folder)
            os.makedirs(path_folder)
        else:
            exit()

def record_rgbd():
    make_clean_folder("../data/realsense/")

    pipeline = rs.pipeline()

    config = rs.config()
    config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

    profile = pipeline.start(config)

    depth_sensor = profile.get_device().first_depth_sensor()
    depth_sensor.set_option(
        rs.option.visual_preset, 3
    )  # Set high accuracy for depth sensor
    depth_scale = depth_sensor.get_depth_scale()

    clipping_distance_in_meters = 1
    clipping_distance = clipping_distance_in_meters / depth_scale

    align_to = rs.stream.color
    align = rs.align(align_to)

    try:
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)
        aligned_depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()

        if not aligned_depth_frame or not color_frame:
            raise RuntimeError("Could not acquire depth or color frames.")

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        grey_color = 153
        depth_image_3d = np.dstack(
            (depth_image, depth_image, depth_image)
        )  # Depth image is 1 channel, color is 3 channels
        bg_removed = np.where(
            (depth_image_3d > clipping_distance) | (depth_image_3d <= 0),
            grey_color,
            color_image,
        )

        color_image = color_image[..., ::-1]

        imageio.imwrite("../data/realsense/depth.png", depth_image)
        imageio.imwrite("../data/realsense/rgb.png", color_image)

    finally:
        pipeline.stop()

    return color_image, depth_image

if __name__ == "__main__":
    record_rgbd()
```

RealSenseCustomerSupport, 2019-04-22:
----------------------------------------------

HI kongsgard,

I tried your code and has no issue on saving depth image. It is just grey level image file and hard to save details.
But if you copy-paste the displaying part of code from opensv_viewer_example.py file (https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py), it would display out the results visually.

Thanks!

RealSenseCustomerSupport, 2019-04-23:
----------------------------------------------

Hi kongsgard,

There is another thing that you may try is replacing the line for getting depth_image with the following line:
depth_image = np.asanyarray(aligned_depth_frame.get_data())
=>
depth_image = cv2.applyColorMap(cv2.convertScaleAbs(aligned_depth_image), cv2.COLORMAP_RAINBOW)
This should give more color info to the depth image, and help show the depth image is good.

Thanks!

vnt1537, 2019-05-03:
How can I save a depth image array created by depth image to csv or How can I retrieve and see the depth image information from .bag if I want to apply some algorithm to that data.

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

Hi vnt1537,
You may refer to the example of read_bag_example.py:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

Hi kongsgard,
Any update from you? If nothing else is needed, this one will be closed.

Thanks!

kongsgard, 2019-05-07:
Thank you for all the feedback!
This solved the issue for me.

closed 2019-05-07 by kongsgard

karlita101, 2021-03-22:
> Hi kongsgard,
>
> There is another thing that you may try is replacing the line for getting depth_image with the following line:
> depth_image = np.asanyarray(aligned_depth_frame.get_data())
> => [...]

# #While this gives a good visual representation, there would be a loss of depth information. Is there another way to keep the accuracy of the depth information in the png?
#8639