realsenseai/librealsense #3659: C# I have saved .bag file saved. Now how do I get depth raw data?

snmathur4, 2019-04-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| Language                            |  C# | 

### Issue Description
I have setup my code in C# to record the depth data into a .bag file. Now what's the workflow to do stuff with the data?
How do I open this file using C# and raw values out of it? thanks

ogoshen, 2019-04-07:
See the playback section in the [c# cookbook](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/csharp/Documentation/cookbook.md#playback)

RealSenseCustomerSupport, 2019-04-16:
----------------------------------------------

Hi snmathur4,

Any update on this one? As ogoshen commented, there are some c# examples. Do you still need any other help?

Thanks!

snmathur4, 2019-04-17:
I got it. thank you!

closed 2019-04-17 by snmathur4