realsenseai/librealsense #3661: Difficulty updating librealsense

jkenney9a, 2019-04-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16)}  | 
| Kernel Version (Linux Only)    |  (4.15)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|                  | 

### Issue Description
I'm having difficulty getting/checking that I'm using the most updated version of librealsense. I originally installed librealsense 2.17.1 and would like to update to the latest version to make use of the updated rs-convert utility. However, after following the instructions for [updating on linux](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md), I'm having difficulty telling if the rs-convert utility has been updated. When I open the realsense-viewer, on the bottom it tells me I'm using librealsense version: 2.17.1, which tells me that perhaps things have not updated correctly. I've also tried removing all realsense packages and installing again as described near the bottom of the installation instructions, but still have the issue.

I've run `rs-convert --version` and the output is: none

At the end of the day, what I'm trying to get is get the latest rs-convert tool that terminates properly so I can call it from within python to convert some bag files (e.g Issue #3208). The fact that it hangs is an issue. 

Thanks!

dorodnic, 2019-04-02:
Hi @jkenney9a 
Are you building from source or installed via apt-get?

jkenney9a, 2019-04-02:
Installed via apt-get

dorodnic, 2019-04-07:
@ev-mp has updated the documentation, please let us know if this helps

closed 2019-04-11 by dorodnic