realsenseai/librealsense #3662: Which is the left imager in the D415?

khrose, 2019-04-02:
According to Figure 2-6 in the D400 data sheet (https://www.mouser.com/pdfdocs/Intel_D400_Series_Datasheet.pdf), I see 4 imagers. Which one is the left one? I'm curious because the depth origin (technically between the left and right imagers) tends to be at the left one.

[image]
Thanks!

agrunnet, 2019-04-03:
The right most one is the color. The second to the right is the “left imager”. It is called left because you have to think of it as a creature with left and right eyes.

dorodnic, 2019-04-04:
Hopefully resolved

closed 2019-04-04 by dorodnic

khrose, 2019-04-04:
@dorodnic Yes this has been resolved. Thanks!